GreenWaves-Technologies/gap_sdk #256: Issue with the Autotiler Compiler

MattAbrs, 2021-07-22:
Hi,

I'm facing some issues with the Autotiler Compiler.

At first, every projects using your GAPflow works, but for some reason, i had to do a make clean sdk.
Since this command, your examples works but my project doesn't work.
When i try to compile the autotiler model, i have this these errors
`référence indéfinie vers « LoadCNN_Copy_Library »` and `référence indéfinie vers « CNN_3DTensorPermute »`
At the end, i had this message : `model_rules.mk:78: recipe for target 'BUILD_MODEL_SQ8BIT/GenTile' failed`

But i don't understand why it's not working anymore because juste before to remake the sdk, everything worked. (In the past, when i made this command, everything still working...)
Also, i'm using the same rules than yourse to compile the autotiler mode.
I hope you can help me and still available if you have some questions.

Thanks in advance.

Yaooooo, 2021-07-26:
Hi,

Firstly, I suggest you to move to the new version of SDK: v4.0.0

Then, everytime when you update your SDK, please do a "make clean" to remove previous build and some old version files (like lib autotiler), and rebuild SDK to download a new version. Because autotiler is a prebuild lib, so we will update the lib from our server and if it hasn't been updated, it will create some build issues.

For your issue, it's maybe due to the new version of autotiler has some reorg, I suggest you move to v4.0.0, then check if your build process (Makefiles), compares to our examples.

Yao

MattAbrs, 2021-07-26:
i resolved my issue thanks to you.

Thanks for your help.

closed 2021-07-26 by MattAbrs