GreenWaves-Technologies/gap_sdk #258: NNTool : Permission not granted.

nbaddi, 2021-07-26:
Hello,

I'm trying to run the gap8_project in occupancy_management,
and I got the following message :

```
Building GAP8 mode with 16 bit quantization
Platform is GVSOC
cd SSDParamsGenerator && make all run
make[1] : on entre dans le répertoire « /home/nbaddi/repo_directory/embedded-os/occupancy_management/gap8_project/SSDParamsGenerator »
make[1]: rien à faire pour « all ».
./genSSDparams

########################################################################
               Generating SSD Anchor Layers Parameters for GAP
########################################################################

...Done
make[1] : on quitte le répertoire « /home/nbaddi/repo_directory/embedded-os/occupancy_management/gap8_project/SSDParamsGenerator »
cp model/lynred.tflite BUILD_MODEL_16BIT
nntool -s model/nntool_script16 BUILD_MODEL_16BIT/lynred.tflite
bash: /home/nbaddi/repo_directory/embedded-os/gap_sdk/tools/nntool/nntool: Permission non accordée
model_rules.mk:41: recipe for target 'BUILD_MODEL_16BIT/lynred.json' failed
make: *** [BUILD_MODEL_16BIT/lynred.json] Error 126

```
The sdk version is 3.9_dev
platform : Ubuntu 18.04.5 LTS

PS : I already installed the nntool requirements listed here: 
https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/nntool/requirements.txt
and build the sdk by running `make clean sdk` in the gap_sdk folder

Thank you in advance for your help,
Have a nice day.

Yaooooo, 2021-07-26:
Hi,

Can you provide more information about your environment:

- How you clone the SDK? (copied from somewhere or direct clone?)
- How you install the SDK (when you running `make clean sdk` ), you are in "root"?
- Can you show us the rights of this file: `ls -all /home/nbaddi/repo_directory/embedded-os/gap_sdk/tools/nntool/nntool`

Yao

Yaooooo, 2021-08-04:
Hi,

I just found one issue related to this, and pushed a fix on master. Please feel free to update it and give a try. 
It should resolved your issue.

If not, please feel free to reopen the issue.

closed 2021-08-04 by Yaooooo