GreenWaves-Technologies/gap_sdk #261: nntool onnx pow ops gives NoneType obj AttributeError

pecanjk, 2021-08-06:
` File "/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 54, in create_graph
    G = self._import_onnx_model(G, model.graph, opset_import, opts)
  File "/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 68, in _import_onnx_model
    self._import_nodes(G, graph, handlers, all_nodes, outputs, opts)
  File "/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 180, in _import_nodes
    params = handler.handle(OnnxNode(node), all_nodes=all_nodes, vars_dict=vars_dict,
  File "/gap_sdk/tools/nntool/importer/onnx/handlers/handler.py", line 61, in handle
    return ver_handle(node, **kwargs)
  File "/gap_sdk/tools/nntool/importer/onnx/handlers/backend/pow.py", line 44, in version_12
    return cls._common(node, **kwargs)
  File "/gap_sdk/tools/nntool/importer/onnx/handlers/backend/pow.py", line 29, in _common
    return super(Pow, cls)._common(node,
  File "/gap_sdk/tools/nntool/importer/onnx/handlers/backend/math_mixin.py", line 76, in _common
    G.add_edge(NNEdge(from_node=inp[0], to_node=params, from_idx=inp[1], to_idx=idx))
  File "/gap_sdk/tools/nntool/utils/graph.py", line 254, in add_edge
    elif edge.to_node.name not in self._nodes:
AttributeError: 'NoneType' object has no attribute 'name'
EXCEPTION of type 'AttributeError' occurred with message: ''NoneType' object has no attribute 'name''
`

pecanjk, 2021-08-24:
Is there anyone who met same problems to handle Pow Ops when converting onnx object detection model at head part

sousoux, 2021-09-20:
Any chance that we can have a model that exhibits this problem?

closed 2021-09-20 by sousoux

pecanjk, 2021-09-22:
> 
> 
> Any chance that we can have a model that exhibits this problem?

please take a look at this model. Thanks.
"https://pan.baidu.com/s/1eIWXtb7uk51moOhIHCkmew"
"verify_code: wx5w"

This model has more problems including 4d transform, pow ops NoneType, resize/reshape and pad. The resize and pad problems fixed in the PR  #264. I can not figure out the pow ops NoneType.

sousoux, 2021-11-17:
I was not able to download this file.