GreenWaves-Technologies/gap_sdk #266: make: 'flash' is up to date.

Manucar, 2021-08-30:
Hi! I am trying to flash the AIdeck board with FreeRTOS with the following command:

``` 
sudo docker run --rm -it -v $PWD:/module/data/ --device /dev/ttyUSB0 --privileged -P bitcraze/aideck /bin/bash -c 'export GAPY_OPENOCD_CABLE=interface/ftdi/olimex-arm-usb-tiny-h.cfg; source /gap_sdk/configs/ai_deck.sh; cd /module/data/;  make clean all image flash'
```
but the output for every example using _PMSIS_OS=freertos_ in the makefile is:

```
rm -f /module/data/BUILD/GAP8_V2/GCC_RISCV/flash.img
    CC main.c
    ASM  gap8_iet.S
    ASM  startup_gap8.S
    ASM  cluster_core.S
    ASM  port_asm.S
    CC  gap8_it.c
    CC  system_gap8.c
    CC  gap_io.c
    CC  string.c
    CC  stdlib.c
    CC  errno.c
    CC  cl_malloc.c
    CC  cl_to_fc_delegate.c
    CC  fc_to_cl_delegate.c
    CC  cl_team.c
    CC  hyperbus_cl_internal.c
    CC  uart_cl_internal.c
    CC  cl_dma_irq.c
    CC  fc_event.c
    CC  fll.c
    CC  gpio.c
    CC  pad.c
    CC  pmu.c
    CC  pmu_internal.c
    CC  pwm.c
    CC  pwm_internal.c
    CC  rtc.c
    CC  rtc_internal.c
    CC  timer.c
    CC  hyperbus.c
    CC  hyperbus_internal.c
    CC  cpi.c
    CC  cpi_internal.c
    CC  dmacpy.c
    CC  dmacpy_internal.c
    CC  i2c.c
    CC  i2c_internal.c
    CC  i2s.c
    CC  i2s_internal.c
    CC  spi.c
    CC  spi_internal.c
    CC  uart.c
    CC  uart_internal.c
    CC  perf_internal.c
    CC  flash.c
    CC  partition.c
    CC  flash_partition.c
    CC  md5.c
    CC  fs.c
    CC  lfs.c
    CC  lfs_util.c
    CC  pi_lfs.c
    CC  read_fs.c
    CC  semihost.c
    CC  host_fs.c
    CC  ota.c
    CC  ota_utility.c
    CC  updater.c
    CC  bootloader_utility.c
    CC  ai_deck.c
    CC  camera.c
    CC  himax.c
    CC  hyperflash.c
    CC  transport.c
    CC  nina_w10.c
    CC  hyperram.c
    CC  spiram.c
    CC  spiflash.c
/tmp/cc2o4CPw.s: Assembler messages:
/tmp/cc2o4CPw.s:2751: Warning: ignoring changed section attributes for .data
    CC  ram.c
    CC  alloc_extern.c
    CC  pi_log.c
    CC  event_kernel.c
    CC  cl_l1_malloc.c
    CC  fc_l1_malloc.c
    CC  l2_malloc.c
    CC  malloc_external.c
    CC  malloc_internal.c
    CC  pi_malloc.c
    CC  mem_slab.c
    CC  device.c
    CC  pmsis_task.c
    CC  pmsis_backend_native_task_api.c
    CC  port.c
    CC  printf.c
    CC  list.c
    CC  queue.c
    CC  tasks.c
    CC  timers.c
    CC  event_groups.c
    CC  stream_buffer.c
    CC  FreeRTOS_util.c
gapy --target=ai_deck --platform=board --work-dir=/module/data/BUILD/GAP8_V2/GCC_RISCV --config-ini=/module/data/config.ini    run --image --binary=/module/data/BUILD/GAP8_V2/GCC_RISCV/test 
gapy --target=ai_deck --platform=board --work-dir=/module/data/BUILD/GAP8_V2/GCC_RISCV --config-ini=/module/data/config.ini    run --flash --binary=/module/data/BUILD/GAP8_V2/GCC_RISCV/test 
gapy --target=ai_deck  gen_flash_image --boot-loader /module/data/BUILD/GAP8_V2/GCC_RISCV/test  -o /module/data/BUILD/GAP8_V2/GCC_RISCV/flash.img 
make: 'image' is up to date.
make: 'flash' is up to date.
```

instead with the run command:
```
sudo docker run --rm -it -v $PWD:/module/data/ --device /dev/ttyUSB0 --privileged -P bitcraze/aideck /bin/bash -c 'export GAPY_OPENOCD_CABLE=interface/ftdi/olimex-arm-usb-tiny-h.cfg; source /gap_sdk/configs/ai_deck.sh; cd /module/data/;  make clean all run'
```
they work correctly. Any idea why?

I am using the docker image provided by the [ Bitcraze team](https://github.com/bitcraze/AIdeck_examples/blob/master/docs/getting-started/docker-gap8.md)

Thank you for your support!

Manuel

sivasivarajah, 2021-09-01:
Hi @Manucar ,

In the first command you used : `make clean all image flash`. The last two targets are implicitly executed when calling `all`. `All` does those three steps : build objects and binary, create image to load on target board and finally load image on target.
In your second command, you do not have it : `make clean all run`.

Manucar, 2021-09-01:
Hi @sivasivarajah, 

I was following this [guide from bitcraze](https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/getting-started/docker-gap8/), and they suggest to use `make clean all image flash` for permanently flash the AIdeck. Sorry but I didn't understand your point, you said `all` automatically load on target, but if I try to execute `make all`, it gives:

```
root@8f3425223fad:/module/data/camlin-thesis/GAP8_to_NINA# make all
gapy --target=ai_deck --platform=board --work-dir=/module/data/camlin-thesis/GAP8_to_NINA/BUILD/GAP8_V2/GCC_RISCV --config-ini=/module/data/camlin-thesis/GAP8_to_NINA/config.ini    run --image --binary=/module/data/camlin-thesis/GAP8_to_NINA/BUILD/GAP8_V2/GCC_RISCV/test 
gapy --target=ai_deck --platform=board --work-dir=/module/data/camlin-thesis/GAP8_to_NINA/BUILD/GAP8_V2/GCC_RISCV --config-ini=/module/data/camlin-thesis/GAP8_to_NINA/config.ini    run --flash --binary=/module/data/camlin-thesis/GAP8_to_NINA/BUILD/GAP8_V2/GCC_RISCV/test 
```
and the board is not flash nor loaded in RAM. So what exact command should I use to _permanently_ flash the AIdeck board? (I am using FreeRTOS in my code).

Yaooooo, 2021-09-01:
Just to be more precise:
image, flash, all are 3 targets in makefile, you can call them separately
or put them together.
Normally, if you are using a filesystem (like READFS_FILES ..... in your
makefile), then when you call "make all", it will call "image and flash"
targets automatically.
However, if you are not using a filesystem, then you have to call them
manually: make image flash.

For your last point, it's the "run" who plays the role:
- If you want to boot your firmware from flash, then you don't need to do
"make run". However, you need to burn the efuse to choose the boot mode.
- otherwise, you should use "make run", to boot the program from JTAG.

Hope this helps.

Manucar, 2021-09-01:
Hi @Yaooooo! thanks for the clarification on the `all` command.

But my problem is that if I use PULPOS, with the command `make clean all image flash`, the AIdeck board is successfully _permanently_ flashed, but instead if I use _PMSIS_OS=freertos_, it gives me the previous output:

```
.....
    CC  pmsis_task.c
    CC  pmsis_backend_native_task_api.c
    CC  port.c
    CC  printf.c
    CC  list.c
    CC  queue.c
    CC  tasks.c
    CC  timers.c
    CC  event_groups.c
    CC  stream_buffer.c
    CC  FreeRTOS_util.c
gapy --target=ai_deck --platform=board --work-dir=/module/data/BUILD/GAP8_V2/GCC_RISCV --config-ini=/module/data/config.ini    run --image --binary=/module/data/BUILD/GAP8_V2/GCC_RISCV/test 
gapy --target=ai_deck --platform=board --work-dir=/module/data/BUILD/GAP8_V2/GCC_RISCV --config-ini=/module/data/config.ini    run --flash --binary=/module/data/BUILD/GAP8_V2/GCC_RISCV/test 
gapy --target=ai_deck  gen_flash_image --boot-loader /module/data/BUILD/GAP8_V2/GCC_RISCV/test  -o /module/data/BUILD/GAP8_V2/GCC_RISCV/flash.img 
make: 'image' is up to date.
make: 'flash' is up to date.
```
and the board is not even loaded with the image, so I think there is something missing for the flashing with FreeRTOS?

Yaooooo, 2021-09-01:
Ah, I thought it's working for you, because you were saying "they work
correctly...."  Never mind:)
OK, I see the problem, you can quickly fix it with:
1. add "image flash" in .PHONY, last line of
file: rtos/freeRTOS/vendors/gwt/rules/freeRTOS_rules.mk
e.g

.PHONY: clean build all run *image flash* debug disdump clean_app

2. add "--force" in the command gapy of target "flash":
e.g

flash: $(BIN)
     gapy $(GAPY_TARGET_OPT) --platform=$(platform) --work-dir=$(BUILDDIR)
$(config_args) $(gapy_args) run --flash *--force* --binary=$(BIN)
$(runner_args)

This should resolve your problem, we will push it to github in the next
release

Thanks

Manucar, 2021-09-01:
Thank you @Yaooooo! This fixed the problem!

closed 2021-09-01 by Manucar