GreenWaves-Technologies/gap_sdk #269: Callback functions management in the OS.

nbaddi, 2021-09-06:
Hi,

I want to use the `pi_time_wait_on` blocking function to wait for an event on a GPIO.
At the same time, I wrote a callback function on the same GPIO in order to increment the PIR_detection variable used in the  main code, but once the `pi_task_wait_on` function is unblocked, I don't enter in the callback function.

Here is the main code:
```
            pi_task_block(&cb_gpio);
            pi_task_wait_on(&cb_gpio);
            printf("PIR_detection = %d\n", PIR_detection);
            while(PIR_detection) {
                    PIR_detection=0;
                    camCaptureImage(IMAGE_T.buff);
                    }
            }
```
The callback function defintion and declaration:
```
    pi_gpio_e gpio_pir_trigger = PIR_GPIO;
    pi_gpio_notif_e irq_type = PI_GPIO_NOTIF_RISE;
    pi_gpio_flags_e cfg_flags = PI_GPIO_INPUT | PI_GPIO_PULL_DISABLE | PI_GPIO_DRIVE_STRENGTH_LOW;
    pi_task_callback(&cb_gpio, __pi_cb_gpio, (void*)gpio_pir_trigger);
    pi_gpio_pin_configure(&gpio, gpio_pir_trigger, cfg_flags);
    pi_gpio_pin_task_add(&gpio, gpio_pir_trigger, &cb_gpio, irq_type);
    pi_gpio_pin_notif_configure(&gpio, gpio_pir_trigger, irq_type);
```

```
static void __pi_cb_gpio(void* arg)
{
    PIR_detection++;
    pi_gpio_e gpio_pin = (pi_gpio_e) arg;
    uint32_t value = 0;
    pi_gpio_pin_read(&gpio, gpio_pin, &value);
}
```
It seems a bit not logical to me the fact that I can't enter in the callback function once the GPIO is unblocked, I wonder if you have any idea how these two functions are handled in the OS?

I'm using a GAPoc B_2.3 board
The sdk version is 4.0
platform : Ubuntu 20.04.2 LTS

Thank you in advance for your help.

sivasivarajah, 2021-09-09:
Hi @nbaddi,

The call to `pi_task_block(&cb_gpio);` is redefining the event task as blocking task while it was previously a callback task. It has been overridden to be only blocking task.
When the GPIO triggers an IRQ, the event task is executed, that is unblocking the task and not the callback.

You can not have both functions for the same event task.
Though you can still wait on a callback task(with `pi_task_wait_on()`), but this is not efficient(performances and energy consumption).

Best,

nbaddi, 2021-09-10:
Hello,
Thank you for you answer.

I still have some questions,

Can we use two different event tasks on the same gpio?

Why we have to define the callback everytime before the `pi_task_wait_on()`?

Does the function pi_task_block allows us to enter in low power? or we have to use the power management functions in this link :  https://greenwaves-technologies.com/manuals/BUILD/PULP-OS/html/group__PM.html#ga48fc269c41617ba4ffcd88d0203aee5e to do that?

To explain even more what I want to do:
I want to stop the code while waiting for a trigger on a gpio, once the gpio is triggered, I want to enter in a callback function in order to increment a variable before continuing the code.

Thank you in advance for your answer.
best regards,

MattAbrs, 2021-11-08:
Hello,

Sorry to bring back this topic again but I am currently working on the same problem so I would like to have the answers to the questions above.
I am using the same version and gapoc.

Thank you in advance for your answer.

Best regards,