GreenWaves-Technologies/gap_sdk #276: Graph constructor exited with an error

MattAbrs, 2021-09-20:
Hi.

I have a problem with the NN constructor.
I am using the Gapuino V2 and the latest version of the SDK.

On my project, I use two different modes, one with the gapoc_b (I have no problem) and one with the gapuino v2.
When I launch my application with the gapuino, I get the following error: 
```Graph constructor exited with an error```

When I look for the problem, I find that it is in the kernel.c file at the following line:```L2_Memory = (AT_L2_POINTER) AT_L2_ALLOC(0, 94381);```
L2_Memory is equal to 0 and returns an error.

I am checking your issues to see if anyone has had the same problem. I haven't found one.

At first I thought it was because of the .dat file for L3 memory, but no (or I missed something).

Was it because I didn't have enough memory? Was it for some other reason that I didn't notice?

Thanks in advance.

Yaooooo, 2021-09-20:
Hi,

Normally it is because you don't have enough memory rest for the L2
allocator. You can use this api to check: *pi_l2_malloc_dump();*

Don't forget, we have only 512KBytes in L2 memory.
When using malloc and free, please follow the rules of First In Last Out.
Otherwise, you will create some bubbles in the memory, and create issue for
the next allocations.

Yao

Yao Zhang
Greenwaves Technologies

MattAbrs, 2021-09-21:
Thanks for your answer. 

I am trying to use the pi_l2_malloc_dump without success although I am using the pmsis_l2_malloc in my file.

Also, since I used the pmsis_l2_malloc function, I will look to see if I am using the malloc and free functions correctly.

I leave this question open until it is resolved.

Thanks again.

Yaooooo, 2021-09-21:
Ah sorry, I forgot it's for GAP8 and maybe you are using pulp-os (If you
haven't specified the PMSIS_OS when you compile), which hasn't implemed
this api. We will add it.
For now, you can use this api instead:

*rt_user_alloc_dump( rt_alloc_l2() );*

MattAbrs, 2021-09-22:
Thank you for your answer, it helped me a lot. 

I solved my problem with this.

closed 2021-09-22 by MattAbrs

paulxiongfree, 2021-09-28:
Hello @Yaooooo,
I am having the same problem when running *AT_L2_ALLOC()*, I have used your method and figured out that the 512KB L2 memory is not enough for my application. Since I am using the GVSOC as a simulator and I found the memory size is configurable according to [this](https://gvsoc.readthedocs.io/en/latest/configuration.html), should I follow the procedures (e.g. --config-file = xxx.json) to enlarge the L2 memory size?