GreenWaves-Technologies/gap_sdk #285: Compilation Warnings

hovavalon, 2021-11-10:
Hello,
I have recently compiled the repository, and while trying to include its files from my own code I found out that there are some compilation warnings, regarding `printf` formatting.
Can this be please fixed?
I have no problem creating the merge request myself, if you will have no problem approving it.
Thanks!

Yaooooo, 2021-11-14:
Hi @hovavalon 

Thanks a lot for reporting this problem, we will take a look.
Could you copy the warning messages here to help us resolve it?
Don't hesitate to push a merge request, we will take a look. But due to the release process, we cannot just accept is (since it will be very hard to upload to our source repo). However, any help and request will be very helpful and welcome!!!

Regards.
Yao

hovavalon, 2021-11-15:
/home/alon/src/gap_sdk/tools/autotiler_v3/DSP_Libraries/FFT_Library.c:1912:94: warning: format ‘%f’ expects argument of type ‘double’, but argument 2 has type ‘int’ [-Wformat=]
 1912 |         printf("\nout_rfft = np.array([\n\t"); for (int j=0; j<((N_fft>>1)+1); j++) printf("%f%+fj, ",  RFFT_Out[j-1][0], RFFT_Out[j-1][1]); printf("])\n");
      |                                                                                             ~^          ~~~~~~~~~~~~~~~~
      |                                                                                              |                       |
      |                                                                                              double                  int
      |                                                                                             %d
/home/alon/src/gap_sdk/tools/autotiler_v3/DSP_Libraries/FFT_Library.c:1912:97: warning: format ‘%f’ expects argument of type ‘double’, but argument 3 has type ‘int’ [-Wformat=]
 1912 |         printf("\nout_rfft = np.array([\n\t"); for (int j=0; j<((N_fft>>1)+1); j++) printf("%f%+fj, ",  RFFT_Out[j-1][0], RFFT_Out[j-1][1]); printf("])\n");
      |                                                                                               ~~^                         ~~~~~~~~~~~~~~~~
      |                                                                                                 |                                      |
      |                                                                                                 double                                 int
      |                                                                                               %+d
/home/alon/src/gap_sdk/tools/autotiler_v3/DSP_Libraries/FFT_Library.c: In function ‘IRFFT_DIF_Par_f16’:
/home/alon/src/gap_sdk/tools/autotiler_v3/DSP_Libraries/FFT_Library.c:2147:89: warning: format ‘%f’ expects argument of type ‘double’, but argument 2 has type ‘int’ [-Wformat=]
 2147 |         printf("\nin_ifft = np.array([\n\t"); for (int j=0; j<(N_fft>>1); j++) printf("%f%+fj, ",  RFFT_Out[j][0], RFFT_Out[j][1]); printf("])\n");
      |                                                                                        ~^          ~~~~~~~~~~~~~~
      |                                                                                         |                     |
      |                                                                                         double                int
      |                                                                                        %d
/home/alon/src/gap_sdk/tools/autotiler_v3/DSP_Libraries/FFT_Library.c:2147:92: warning: format ‘%f’ expects argument of type ‘double’, but argument 3 has type ‘int’ [-Wformat=]
 2147 |         printf("\nin_ifft = np.array([\n\t"); for (int j=0; j<(N_fft>>1); j++) printf("%f%+fj, ",  RFFT_Out[j][0], RFFT_Out[j][1]); printf("])\n");
      |                                                                                          ~~^                       ~~~~~~~~~~~~~~
      |                                                                                            |                                  |
      |                                                                                            double                             int
      |                                                                                          %+d
/home/alon/src/gap_sdk/tools/autotiler_v3/DSP_Libraries/FFT_Library.c:2161:90: warning: format ‘%f’ expects argument of type ‘double’, but argument 2 has type ‘int’ [-Wformat=]
 2161 |         printf("\nout_rfft = np.array([\n\t"); for (int j=0; j<(N_fft>>1); j++) printf("%f, %f, ",  RFFT_Out[j][0], RFFT_Out[j][1]); printf("])\n");
      |                                                                                         ~^          ~~~~~~~~~~~~~~
      |                                                                                          |                     |
      |                                                                                          double                int
      |                                                                                         %d
/home/alon/src/gap_sdk/tools/autotiler_v3/DSP_Libraries/FFT_Library.c:2161:94: warning: format ‘%f’ expects argument of type ‘double’, but argument 3 has type ‘int’ [-Wformat=]
 2161 |         printf("\nout_rfft = np.array([\n\t"); for (int j=0; j<(N_fft>>1); j++) printf("%f, %f, ",  RFFT_Out[j][0], RFFT_Out[j][1]); printf("])\n");
      |                                                                                             ~^                      ~~~~~~~~~~~~~~
      |                                                                                              |                                 |
      |                                                                                              double                            int
      |                                                                                             %d
/home/alon/src/gap_sdk/tools/autotiler_v3/DSP_Libraries/FFT_Library.c: In function ‘SwapSamples_Par_f16’:
/home/alon/src/gap_sdk/tools/autotiler_v3/DSP_Libraries/FFT_Library.c:2307:97: warning: format ‘%f’ expects argument of type ‘double’, but argument 2 has type ‘int’ [-Wformat=]
[... cut ...]
  919 |                 printf("out_window_c = np.array([\n\t\t"); for(int i=0; i<FFT_Dim ; i++) printf("%f, ", OutFrame[i]); printf("])\n");
      |                                                                                                  ~^     ~~~~~~~~~~~
      |                                                                                                   |             |
      |                                                                                                   double        int
      |                                                                                                  %d