GreenWaves-Technologies/gap_sdk #290: Can't process tflite models in nntool

plinnie, 2021-12-06:
I have a non-quantized version, and a quantized version of a model, and both give me issues when attempting to load it into the GAP SDK. I cloned the latest master this weekend. The model is loosly based on MobilnetV3. I'm using Python 3.9 and Tensorflow 2.7

The non-quanitzed model
--------------------------

Written as follows:
```python
    converter = tf.lite.TFLiteConverter.from_keras_model(model)
    tflite_model = converter.convert()
    tflite_file.write_bytes(tflite_model)
```

And then attempted to process as follows:
```bash
(py39tf27) vincent@beult:~/projects/other/gap_sdk/tools/nntool$ ./nntool ~/data/model.tflite
Welcome to NNTOOL. Type help or ? to list commands.

settings - set log level to INFO
log_level - was: 'INFO'
now: 'INFO'
open - opening graph file /home/vincent/data/hummingbird/model.tflite load_quantization = False
tflite - Importing TFLITE model version 3
strided_slice - reducing STRIDED_SLICE_0_1 to a constant
pack - reducing PACK_0_2 to a constant
strided_slice - reducing STRIDED_SLICE_0_11 to a constant
pack - reducing PACK_0_12 to a constant
strided_slice - reducing STRIDED_SLICE_0_32 to a constant
pack - reducing PACK_0_33 to a constant
strided_slice - reducing STRIDED_SLICE_0_44 to a constant
pack - reducing PACK_0_45 to a constant
clean_dangling_nodes - removing node STRIDED_SLICE_0_1 which is not connected to an output
clean_dangling_nodes - removing node STRIDED_SLICE_0_11 which is not connected to an output
clean_dangling_nodes - removing node STRIDED_SLICE_0_32 which is not connected to an output
clean_dangling_nodes - removing node STRIDED_SLICE_0_44 which is not connected to an output
clean_dangling_nodes - removing node SHAPE_0_0 which is not connected to an output
clean_dangling_nodes - removing node SHAPE_0_10 which is not connected to an output
clean_dangling_nodes - removing node SHAPE_0_31 which is not connected to an output
clean_dangling_nodes - removing node SHAPE_0_43 which is not connected to an output
duplicate_constants - node mn3d1activationactivationmuly has more than one out edge and will be duplicated
nngraph - update graph dimensions
set_aliases - looking for aliased edges
nngraph - calculate liveness
nngraph - update graph dimensions
set_aliases - looking for aliased edges
nngraph - calculate liveness
nngraph - update graph dimensions
set_aliases - looking for aliased edges
nngraph - calculate liveness
fuse_pad - adding padding from: PAD_0_4 to filter: CONV_2D_0_5 - has 0 reshapes
fuse_pad - adding padding from: PAD_0_7 to filter: DEPTHWISE_CONV_2D_0_8 - has 0 reshapes
fuse_pad - adding padding from: PAD_0_23 to filter: DEPTHWISE_CONV_2D_0_24 - has 0 reshapes
nngraph - update graph dimensions
set_aliases - looking for aliased edges
nngraph - calculate liveness
(NNT model.tflite 0) adjust
adjust_order - adding transposes to correct tensor order for AT kernels
global_pool - global pool fusion MEAN_0_9: inserting transpose before operation
global_pool - global pool fusion MEAN_0_30: inserting transpose before operation
global_pool - global pool fusion MEAN_0_42: inserting transpose before operation
set_aliases - looking for aliased edges
eliminate_transposes - eliminating unnecessary transposes
eliminate_transposes - search for transposes
eliminate_transposes - trying to eliminate CONV_2D_0_5_trans_in0 upwards
eliminate_transposes - pass reshape RESHAPE_0_3 up trans: old (2, 0, 1) new (0, 1) shape: old 186x62 new 186x62x1
eliminate_transposes - rewrite reshape to 186x62->1x186x62
eliminate_transposes - accepted input_1 - input without fixed order - transpose input (0, 1)
eliminate_transposes - trying to eliminate CONV_2D_0_5_trans_in0 downwards
eliminate_transposes - rejected CONV_2D_0_5  - sensitive to order - inserting transpose (2, 0, 1)
eliminate_transposes - found elimination for CONV_2D_0_5_trans_in0 upwards - 1 eliminated
eliminate_transposes - trying to eliminate CONV_2D_0_5_trans_out0 upwards
[... 522 lines cut ...]
    return func(cmd2_app, args)
  File "/home/vincent/projects/other/gap_sdk/tools/nntool/interpreter/commands/open.py", line 146, in do_open
    self.__open_graph(args)
  File "/home/vincent/projects/other/gap_sdk/tools/nntool/interpreter/commands/open.py", line 115, in __open_graph
    G = create_graph(graph_file, opts=opts)
  File "/home/vincent/projects/other/gap_sdk/tools/nntool/importer/importer.py", line 53, in create_graph
    graph = importer.create_graph(filename, opts)
  File "/home/vincent/projects/other/gap_sdk/tools/nntool/importer/tflite2/tflite.py", line 137, in create_graph
    G.quantization = quantizer.quantize()
  File "/home/vincent/projects/other/gap_sdk/tools/nntool/quantization/quantizer/new_quantizer.py", line 882, in quantize
    self.bfs_pass()
  File "/home/vincent/projects/other/gap_sdk/tools/nntool/quantization/quantizer/new_quantizer.py", line 569, in bfs_pass
    self._bfs_pass(self._G, node, self._qset, visited)
  File "/home/vincent/projects/other/gap_sdk/tools/nntool/quantization/quantizer/new_quantizer.py", line 489, in _bfs_pass
    qrec = self._resolve_qrec(
  File "/home/vincent/projects/other/gap_sdk/tools/nntool/quantization/quantizer/new_quantizer.py", line 544, in _resolve_qrec
    qrec = self._call_quantizer(cur_G,
  File "/home/vincent/projects/other/gap_sdk/tools/nntool/quantization/quantizer/new_quantizer.py", line 384, in _call_quantizer
    raise MissingStatsError(
quantization.quantizer.new_quantizer.MissingStatsError: handler type requires activation stats to be present
EXCEPTION of type 'MissingStatsError' occurred with message: 'handler type requires activation stats to be present'
```
I'm new with using nntool and the GAP SDK,so manybe I missed some important step or limitations. Some guidence would be welcome. 

Both models can be downloaded from here: https://surfdrive.surf.nl/files/index.php/s/9m2EjoerKtJHnBk

sousoux, 2021-12-06:
Thanks for the report. Definitely an issue there.

sousoux, 2021-12-07:
This is fixed internally and will be in the next release in the next few days.

sousoux, 2021-12-08:
FYI. Fix is in new version pushed today. Please let us know if this fixes your issue.

plinnie, 2021-12-08:
Yes, I can load and process both models now. Thanks!

closed 2021-12-08 by plinnie