GreenWaves-Technologies/gap_sdk #292: 8bit Error: ProcessGraphEdges, Graph local argument declared, written but not read

pecanjk, 2021-12-10:
when I use 8bit model. there have such problem for sdk_4.8. while previously in sdk_4.0.0. there is no such problem
Here is the log
```
==== Process Tiling For User Kernel:                 S44_Conv2d_18x64x1x1 =======================
S44_Conv2d_18x64x1x1 Partition[0] Size =   7731 (Min:   1024, Max:  19749), Fraction:       1.00, Giving:  48736 Bytes out of  48736 Bytes

Reference object:                  In2, Dim=80
	                 In2 Dim:  80, TileOverlap:  0, Ratio: 1.000000
	            TransIn2 Dim:  80, TileOverlap:  0, Ratio: 1.000000
	                 Out Dim:  80, TileOverlap:  0, Ratio: 1.000000

Kernel: S44_Conv2d_18x64x1x1, Total Raw Memory: 12956 fits into L1 memory 48736. Promoting all kernel arguments to initialized buffers.
Ker: S44_Conv2d_18x64x1x1, Arg:        In1, Size:   1152, Base1:      0, Base2:      0
Ker: S44_Conv2d_18x64x1x1, Arg:        In2, Size:   5120, Base1:   1152, Base2:      0
Ker: S44_Conv2d_18x64x1x1, Arg:   TransIn2, Size:   5120, Base1:   6272, Base2:      0
Ker: S44_Conv2d_18x64x1x1, Arg:       Bias, Size:     72, Base1:  11392, Base2:      0
Ker: S44_Conv2d_18x64x1x1, Arg:        Out, Size:   1440, Base1:  11464, Base2:      0
Ker: S44_Conv2d_18x64x1x1, Arg:      Scale, Size:     20, Base1:  12904, Base2:      0
Ker: S44_Conv2d_18x64x1x1, Arg:     ScaleN, Size:     20, Base1:  12924, Base2:      0
Ker: S44_Conv2d_18x64x1x1, Arg:      Infos, Size:     12, Base1:  12944, Base2:      0
S44_Conv2d_18x64x1x1 For Iter Space: 0 Iteration count:   1, Given L1 Memory:  48736, Used L1 Memory:  12956, Reusable Memory: 35780, Used L2 Memory: 0
=================================================================================================

Graph Node S37_Op_RESIZE_NEAREST_NEIGHBOR_0_46, Stacked tensor S38_Output replaces source binding S37_Output in Out
Graph Node S16_MaxPool_2x2, Stacked tensor S38_Output replaces source binding S15_Output in In
Graph Node S15_Conv2d_64x32x3x3_Relu, Stacked tensor S38_Output replaces source binding S15_Output in Out
Error: ProcessGraphEdges, Graph local argument declared, written but not read S19_Output
Execution aborted

```

sousoux, 2021-12-17:
Is there any chance of getting the graph to test please? Please could you use a service that does not require registration to transfer it.

sousoux, 2021-12-24:
No activity

closed 2021-12-24 by sousoux