GreenWaves-Technologies/gap_sdk #294: Network fails to build / execute with input formatter

gemenerik, 2021-12-17:
I have a retrained MobileNetV2 with a custom head.
I want to add an input formatter to my graph, to transpose RGB888 HxWxC images to RGB888 CxHxW.

In my nntool script I have tried adding the input formatter (`imageformat input_1 rgb888 shift_int8`) both _before_ and _after_ the `adjust` command. Based on this, nntool behaves differently.

1. Adding it before `adjust`, nntool gets stuck on `save_state` with `three_d_transpose_kernels_generator - generating for transpose in 3x120x120 out 120x3x120 trans (2, 0, 1)`. The out resolution is curious...

2. Adding after `adjust` tiles the graph. The generated kernels now take an `unsigned char *` as input, as expected. However, running it on-board results in a freeze upon execution of the network (`pi_cluster_send_task_to_cl`), on GVSOC I get the following error upon execution: 

    ```
    [/sys/board/chip/cluster/pe0/trace] SP-based access outside stack (addr: 0x1000beb0, stack_start: 0x1000beb8, stack_end: 0x1000ce74)
    make: *** [/home/rik/gap_sdk_480/utils/rules/pulp_rules.mk:252: run] Aborted (core dumped)
    ```

* Tried both `shift_int8` and `offset_int8`.
* In all cases, showing the graph in nntool shows the added input formatter is added as expected.
    ```
    +------+--------------------------+-------------------------+--------------+-------------+--------+--------+---------+---------+-------------------------+-------------------------+
    |  1   | input_1_formatter        | image_format            |  120x120x3   |  3x120x120  |  0/0   | 86400  |    0    |         | FORMAT_CHANGE Fmt:      | in: hxwxc out: cxhxw    |
    |      |                          |                         |              |             |        |        |         |         | RGB888 Norm: SHIFT_INT8 |                         |
    +------+--------------------------+-------------------------+--------------+-------------+--------+--------+---------+---------+-------------------------+-------------------------+
    ```
* Tried both a prequantized and unquantized (quantized in nntool) TensorFlow Lite model.
* I managed to run the network both on GVSOC and on-board without the input formatter.
* Running GAP SDK 4.8.0

[classification_tflite.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/7734414/classification_tflite.zip)

gemenerik, 2021-12-17:
Turns out I'm getting 100% CPU usage on a single thread for case 1. Perhaps I was too impatient, I'll give it some time to run.

gemenerik, 2021-12-17:
No visible progress after 15 minutes.

sousoux, 2021-12-24:
Please pull latest SDK and try again

gemenerik, 2022-03-15:
Did not verify on the same net as the issue, but managed to implement input formatters into my networks succesfully with the updated SDK

closed 2022-03-15 by gemenerik