GreenWaves-Technologies/gap_sdk #295: Protecting variables used by callbacks

MattAbrs, 2021-12-20:
Hi, 

I use Gapoc_b v2 and SDK version 4.0.

I am trying to count the number of PIR rise events. To do this, I have a callback function to increment the counter I use.
However, I also write 0 to this counter when I have finished what I want to do.
Is there a function to protect the variable to avoid conflict? Because I am writing to this variable in two places in the code. At the moment I am using the pi_gpio_callback_remove function before setting the variable to 0, is this a good way to do it?

Thanks in advance.

Yaooooo, 2021-12-20:
Hi,

I am not sure that I understood what you are going to do. To be clear:

You would like to do something like this:
```
volatile int counter = 0;
void callback_pir()
{
counter++;
}

void end_of_count()
{
counter = 0;
}

void main()
{
............
// set the GPIO with callback based on IRQ on rising edge

while (counter < 1000) // for example
{
   // Do something
}

// PIR tirggered 1000 times, End
end_of_counter();
}

```

Am I right? If that's what you mean, I don't see the problem, and even no need `volatile`, why you want/need to protect it? 

Regards.

MattAbrs, 2021-12-20:
Yes, that's what I do. Except that in the while condition, I have a function to check if the PIR has been triggered (I check if the counter at time t and the counter at time t-1 is different).

I thought we might have a conflict if I was in the end_of_counter function to set the counter to 0 and, at the same time, the callback was called, and set the counter ++. That's why I want to protect it.

Regards.

Yaooooo, 2021-12-20:
OK, so before the end_of_counter, you are not planning to close the GPIO (PIR)? 

In this case, you can add this in the end of counter:
```
void end_of_count()
{
int irq = pi_irq_disable();
counter = 0;
pi_irq_restore(irq);
}

```
In this case, the end_of_counter is protected from the callback.

MattAbrs, 2021-12-20:
Thank you for all the information you have given me, it has helped me a lot.

Best regards.

closed 2021-12-20 by MattAbrs