GreenWaves-Technologies/gap_sdk #297: make minimal error

Nagatejapallapu, 2021-12-21:
make[4]: /usr/lib/gap_riscv_toolchain/bin/riscv32-unknown-elf-gcc: Command not found
mk/rules.mk:57: recipe for target '/home/gap/gap_sdk/build/gap/rt/cl/kernel/sync_mc.o' failed
make[4]: *** [/home/gap/gap_sdk/build/gap/rt/cl/kernel/sync_mc.o] Error 127
make[4]: Leaving directory '/home/gap/gap_sdk/gap8/rtos/pulp/pulp-os'
gap_sdk.mk:74: recipe for target 'install_rt_gap8' failed
make[3]: *** [install_rt_gap8] Error 2
make[3]: Leaving directory '/home/gap/gap_sdk/gap8/rtos/pulp/pulp-os'
Makefile:49: recipe for target 'build.pulprt' failed
make[2]: *** [build.pulprt] Error 2
make[2]: Leaving directory '/home/gap/gap_sdk/gap8/rtos/pulp'
gap8/gap8.mk:114: recipe for target 'pulp-os' failed
make[1]: *** [pulp-os] Error 2
make[1]: Leaving directory '/home/gap/gap_sdk'
Makefile:42: recipe for target 'all' failed
make: *** [all] Error 2

gemenerik, 2021-12-21:
It looks like you haven't properly installed the [GAP/RISC-V toolchain](https://github.com/GreenWaves-Technologies/gap_sdk#download-and-install-the-toolchain).

Nagatejapallapu, 2021-12-21:
Enter the path where the compiler will be installed (you may need to execute this script with sudo): /usr/lib/gap_riscv_toolchain
Install path: /usr/lib/gap_riscv_toolchain
mkdir: cannot create directory ‘/usr/lib/gap_riscv_toolchain’: Permission denied

gemenerik, 2021-12-21:
_"you may need to execute this script with sudo"_

So; `sudo ./install.sh`

Nagatejapallapu, 2021-12-21:
ImportError: No module named configparser:
  File "/home/gap/gap_sdk/gap8/rtos/pulp/archi_pulp/SConstruct", line 4:
    import pulp_config as plpconfig
  File "/home/gap/gap_sdk/utils/gap_configs/python/pulp_config.py", line 19:
    import json_tools as js
  File "/home/gap/gap_sdk/utils/gapy/json_tools.py", line 23:
    import configparser
Makefile:2: recipe for target 'build' failed
make[3]: *** [build] Error 2
make[3]: Leaving directory '/home/gap/gap_sdk/gap8/rtos/pulp/archi_pulp'
Makefile:7: recipe for target 'do.archi' failed
make[2]: *** [do.archi] Error 2
make[2]: Leaving directory '/home/gap/gap_sdk/gap8/rtos/pulp'
gap8/gap8.mk:114: recipe for target 'pulp-os' failed
make[1]: *** [pulp-os] Error 2
make[1]: Leaving directory '/home/gap/gap_sdk'
Makefile:42: recipe for target 'all' failed
make: *** [all] Error 2

Nagatejapallapu, 2021-12-22:
Registration is required to load the GAP8 AutoTiler library

In case you have already registered, you can directly enter the link
to the AutoTiler library that you received from an email.
Otherwise just press Enter.
Enter URL from email: 
You will be prompted for your name, company and email address and the
link for the AutoTiler libray will be sent to your email address.
This information is used uniquely to keep track of AutoTiler users.
Enter your first name:

gemenerik, 2021-12-22:
Read

Nagatejapallapu, 2021-12-22:
how to register for auto tiler, 

Registration is required to load the GAP8 AutoTiler library

In case you have already registered, you can directly enter the link
to the AutoTiler library that you received from an email.
Otherwise just press Enter.
Enter URL from email: 
You will be prompted for your name, company and email address and the
link for the AutoTiler libray will be sent to your email address.
This information is used uniquely to keep track of AutoTiler users.
Enter your first name: 
Enter your last name: 
Enter your company name: 
Enter your country: india
Country  IN
Enter your email address:           
Please enter a valid email address
Enter your email address:

gemenerik, 2021-12-22:
Enter an email address

Nagatejapallapu, 2021-12-22:
python3 get_tiler.py
Registration is required to load the GAP8 AutoTiler library

In case you have already registered, you can directly enter the link
to the AutoTiler library that you received from an email.
Otherwise just press Enter.
Enter URL from email: 
You will be prompted for your name, company and email address and the
link for the AutoTiler libray will be sent to your email address.
This information is used uniquely to keep track of AutoTiler users.
Enter your first name: 
Enter your last name: 
Enter your company name: 
Enter your country: india
Country  IN
Enter your email address:           
Please enter a valid email address
Enter your email address: nagatejapallapu@gmail.com
Triggering email ... please wait
FATAL ERROR: HTTPSConnectionPool(host='hooks.zapier.com', port=443): Max retries exceeded with url: /hooks/catch/2624512/e6qico/ (Caused by SSLError(SSLError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)'),))
Makefile:17: recipe for target '/home/Gap/gap_sdk/.tiler_url' failed
make[2]: *** [/home/Gap/gap_sdk/.tiler_url] Error 1
make[2]: Leaving directory '/home/Gap/gap_sdk/tools/autotiler_v3'
gap8/gap8.mk:258: recipe for target 'autotiler' failed
make[1]: *** [autotiler] Error 2
make[1]: Leaving directory '/home/Gap/gap_sdk'
Makefile:42: recipe for target 'all' failed
make: *** [all] Error 2

Nagatejapallapu, 2021-12-22:
an error is occur

justanotheruserontheweb, 2024-03-14:
I know this is an old thread, but Nagatejapallapu is currently going around message boards asking people how to hack into phones.

I created this account just to let you all know.