GreenWaves-Technologies/gap_sdk #305: sdk build, flatbuffers

berab, 2022-01-22:
Hi,

While bulding the SDK (running make sdk), I am getting the following Error:

`CMake Error: The source directory "/home/kilic/Documents/gap_sdk/tools/nntool/build/packages/flatbuffers/build" does not exist.  `
`Specify --help for usage, or press the help button on the CMake GUI.  `
`Makefile:224: recipe for target 'cmake_check_build_system' failed  `
`make[2]: *** [cmake_check_build_system] Error 1  `
`make[2]: Leaving directory '/home/kilic/Documents/gap_sdk/tools/nntool'  `
`gap8/gap8.mk:202: recipe for target 'nntool' failed  `
`make[1]: *** [nntool] Error 2  `
`make[1]: Leaving directory '/home/kilic/Documents/gap_sdk'  `
`Makefile:42: recipe for target 'all' failed  `
`make: *** [all] Error 2 `

Anyone had the same issue or knows how to handle it?

Best regards,
Beran

berab, 2022-01-22:
In case anyone has the same issue in the future, I could handle the issue by building flatbuffers.

`mkdir build`
`cmake ..`
`make -j`

closed 2022-01-22 by berab

HaolinChen, 2022-03-02:
> In case anyone has the same issue in the future, I could handle the issue by building flatbuffers.
> 
> `mkdir build && cd build` `cmake ..` `make -j`

It works! Thank you!