GreenWaves-Technologies/gap_sdk #307: Advice on debugging `SP-based access outside stack` errors

gemenerik, 2022-01-27:
I have had several CNN architectures giving me the following error when executing in GVSOC:
```
[/sys/board/chip/cluster/pe0/trace] SP-based access outside stack (addr: _, stack_start: _, stack_end: _)
```

And freezing without warning when executing on-board.

I am a bit puzzled where to look for the cause of such errors. I noticed that dropping layers, activations or the number of filters in convolutions can all help resolve the problem, but at the same time I have had different models with more layers and a multitude of the parameters run successfully.

I am inclined to think this is related to the tiling.

Perhaps you could share some information on how to debug this. This could help bug reporting, and at least help me make an informed decision about workarounds.

Pafrak, 2022-01-27:
Hi @gemenerik ,
This is a stack overflow on the PE0 of the cluster, you can increase the stack size and it should solve the issue.

Pafrak, 2022-01-27:
Often in our projects there are three variable to set stack size:

- MAIN_STACK_SIZE which is the stack size of the Fabric Controller (FC)
- STACK_SIZE which is the stack size of the PE0 of the cluster (This is the one you want to increase)
- SLAVE_STACK_SIZE which is the stack size of the PE1-PE7

closed 2022-01-27 by Pafrak

DYX-JL, 2022-08-11:
> 

Hi,
I found that this issue can also be solved by only increasing SLAVE_STACK_SIZE. But I don't know the reason.