GreenWaves-Technologies/gap_sdk #314: Makefile RULES_DIR path

NickWroble, 2022-03-06:
When ever I try to compile the helloworld example, the Makefile fails with 

`Makefile:16: /pmsis_rules.mk: No such file or directory`
`make: *** No rule to make target '/pmsis_rules.mk'.  Stop.`

What I'm confused about is that the Makefile points to pmsis_rules.mk using `$(RULES_DIR)`, which in my case maps to a directory that I have confirmed contains pmsis_rules.mk:

`nickw@L3:~$ find . -name pmsis_rules.mk`
`./gap_riscv_toolchain_ubuntu_18/gap_sdk/install/workstation/rules/pmsis_rules.mk`
`./gap_riscv_toolchain_ubuntu_18/gap_sdk/utils/rules/pmsis_rules.mk`

I'm not quite sure where to go from here, since it seems like Make just isn't seeing that file. 

I found the same issue here: https://github.com/GreenWaves-Technologies/gap_sdk/issues/131 which recommended installing the pyelftools Python library but that's already installed.

NickWroble, 2022-03-06:
Found my issue. I need to source my board each time I recompile.

closed 2022-03-06 by NickWroble

mcp-2, 2022-05-23:
> 

How do you source your board?