GreenWaves-Technologies/gap_sdk #318: Waking up from Retentive Deep Sleep

JamesSam14, 2022-03-30:
Hello everyone. 

I am using Gapuino v1.1 board.  

While running the example included in this repo, the core does not wake up after retentive deep sleep (gap_sdk/examples/pmsis/periph/deep_sleep).

So the execution is halted and after a short period I get the message: 
Error: Burst read failed
can't read halt state

Any ideas? Does this code work on your boards?

Thank you!

haugoug, 2022-04-13:
Hello,
I don't have any board with me to try it, I cannot check is the test is still working well.
At least you can try to recompile your test with uart and get the printf on something like cutecom. I think the default baudrate is 115200. You have to compile with this command: make clean all run io=uart
The error you get is because when the chip go to sleep, the jtag connection is broken and openocd complains, so that's normal.
I see in the code that it should blink a led though so indeed it seems it is never waking-up.
Are you trying on freertos or pulpos ?
Best,
Germain

JamesSam14, 2022-04-20:
Hello Germain,

thank you very much for the reply.

I am using freertos, as it is defined in the makefile provided.
Even if I remove the printfs and try to blink the led it is obviously not
waking up.

Note that both RTC wake up and GPIO wake up fail.

In any case, I appreciate your effort. If you find out anything related to
that let me know.

Best Regards,
Dimitrios

haugoug, 2022-05-18:
Hello Dimitrios,
could try it on pulpos ?
Best regards,
Germain

JamesSam14, 2022-06-02:
Hello Germain,

so I figured out that the sleep functions are only implemented in freertos, not in pulpos.

I cannot find similar functions in pulpos, so I think I will have to check if there is a bug in the freertos implementation.

Thank you again for your time.

Best Regards,
Dimitrios

JamesSam14, 2022-06-03:
Moreover, the register responsible for configuring the sleep mode does not have the correct values for retentive deep sleep.

It seems like the functions called within the program have no influence on it.

JamesSam14, 2022-06-08:
More specifically,
 the register SAFE_PMU_SLEEPCTL, which is responsible for the sleep mode configuration, is both Read and Write (some bits are read-only). Unfortunately, it is not correctly set for retentive deep sleep through the example. If I try to write explicitly on the register at byte level with a C pointer it has no effect (maybe because of the Read-only bits). 
 Is there any other way to explicitly write on this register?

ROMAINPC, 2022-06-10:
Hello,
I have the same issue.
I confirm that deep_sleep feature seems to be not implemented for pulpos, neither does the rtc. Is there a similar feature for pulpos ?
GPIO and RTC 17s alarm are working on my GAPPoc-A with gap_sdk 4.12.0.

However when  it is supposed to wakes up, the UART just prints HelloWorld, no blinking LED. I tried to print as  `%p` variables in `__pi_pmu_boot_state_get()` , they are displayed only at the cold boot, not after the second HelloWorld.

It is the same behavior with `make run` or when booting from flash.

Best regards,
Romain

ROMAINPC, 2022-06-13:
Hello,
Booting from not retentive deep sleep is working on my board.

For the retentive deep sleep, I noticed that the program is freezing on the first printf, so I removed them. But then it's freezing either on the end of `pi_gpio_pin_write()` or on `pi_time_wait_us()`. I just manage to turn on the LED.

Best regards,
Romain

ROMAINPC, 2022-06-15:
Hello,

I found a way to switch my board to deep sleep with Pulp OS.
I used [this library](https://greenwaves-technologies.com/manuals/BUILD/PULP-OS/html/group__PM.html). To doing something similar to the `deep_sleep.c` example there is no need to add more includes than `pmsis.h` and `bsp/bsp.h`.

The code is working with PulpOS. You have to execute this code to enter deep sleep  : 
```
// ... open and config GPIO
pi_gpio_pin_configure(...);

rt_pm_wakeup_gpio_conf(1, PI_GPIO_A3_PAD_11_B4,
                               RT_PM_WAKEUP_GPIO_RISING_EDGE);
rt_pm_wakeup_clear_all(); // important, otherwise board will wakeup immediately 
rt_pm_state_switch(RT_PM_STATE_DEEP_SLEEP, RT_PM_STATE_FAST);
```
Use `rt_pm_wakeup_state()` to know to know which boot you come from.

There is also the mode `RT_PM_STATE_SLEEP` for retentive sleep. I haven't tested in detail yet. It just seems that the program does not stop in `rt_pm_state_switch()`, so it may be necessary to review the flow of execution.

JamesSam14, 2022-06-15:
Hello Romain,

I really appreciate your feedback. I could not find those functions on Pulp OS .

I will give it a try myself as soon as possible.

Thank you!

Best Regards,
Dimitrios

JamesSam14, 2022-06-17:
Hello again,

so I tried to use the library you suggest. 

It seems to be working but its functionality is limited.

So, by observing led blinking behavior it seems like the sleep mode works (program stops at rt_pm_state_switch). Then, it wakes up through external GPIO but continues execution from the start of main, and on top of that, the printf stops working.

It requires more time and measurements of current consumption with an external device to make sure that sleep mode is activated.

Best Regards,
Dimitrios

ROMAINPC, 2022-06-18:
Hello,
I also did some tests, indeed it is necessary to make a divergence of execution using `rt_pm_wakeup_state()`.
I tried a program using the camera. Be aware to reconfigure the GPIO just before entering deep sleep mode.
Also I did not specify that the awakening is a boot, and therefore the card must be flashed with `make image flash`, `make run` can still be used for the first boot in cold mode.

Therefor because deep sleep is a complete shutdown, printf stops working due to the disconnection of the jtag. You have to use `io=uart` at compilation, and connect an uart.

I did'n try normal sleep mode yet but may be it will be possible to have a more complicated  flow of execution than just two differents behaviors. On GAPPoc I have a Hyperflash, so I also can use it to store booleans and add more modes. Although I don't know if it's very good practice.

I measured the blink program. Here are my results on GAPPoc-A v2 :
| OS | deep sleep |  blinking LED |
| --- | --- | --- |
| FreeRTOS | 11mA | 2.5<->5.5mA  |
| Pulp OS | 7.5mA | 9<->12mA |

Note that booting from `make run` instead of standard reset consumes a bit more.