GreenWaves-Technologies/gap_sdk #320: SDK build error for target GAPUINO

JamesSam14, 2022-04-21:
Hello everyone,

I have followed all the instructions line by line but the "make all" fails for GAPUINO.
The error I get is this:
make[3]: Entering directory '/home/samakovl/ESL/PULP/gap_sdk/gap8/rtos/pulp/archi_pulp'
scons
scons: Reading SConscript files ...
b'\x1b[91mFATAL ERROR: \x1b[0mExpecting property name enclosed in double quotes: line 14 column 7 (char 274)\n'
CalledProcessError: Command '['plpfiles', 'copy', '--item=archi_files']' returned non-zero exit status 255.:
  File "/home/samakovl/ESL/PULP/gap_sdk/gap8/rtos/pulp/archi_pulp/SConstruct", line 21:
    for file in subprocess.check_output(shlex.split('plpfiles copy --item=archi_files')).decode('UTF-8').split():
  File "/usr/lib/python3.8/subprocess.py", line 415:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 516:
    raise CalledProcessError(retcode, process.args,

However, when using the target platform GAPUINO_V2 the build is successful and the execution on the board works as expected.

To sum up I have 2 questions:
1. Is there a solution to the above compilation error?
2. How safe is it to use GAPUINO_V2 build for a Gapuino v1.1 board? Even if the results of the execution are correct, is it reliable to evaluate the performance cycles or even the energy consumption of the board?

Thank you for reading this. 

Any feedback appreciated.

Best,
Dimitrios

eguidieri, 2022-08-01:
Hi Dimitrios,

I was having the same problem, so I'm answering to this question for the other poor souls that need to work with a Gapuino RevA.

There is a configuration file with a unneeded trailing comma at line 14 that crash the configuration.

The file is $GAP_SDK/utils/gap_configs/configs/boards/gapuino.json

This is my patch.

diff --git a/utils/gap_configs/configs/boards/gapuino.json b/utils/gap_configs/configs/boards/gapuino.json
index e68c8fe4..3037734e 100644
--- a/utils/gap_configs/configs/boards/gapuino.json
+++ b/utils/gap_configs/configs/boards/gapuino.json
@@ -10,7 +10,7 @@
       },
       "ram": {
         "@include@" : "devices/hyperram.json",
-        "cs": 0,
+        "cs": 0
       },
       "camera": {
         "@include@": "devices/camera.json",

I hope this help,
Errico Guidieri

fernandoFernandeSantos, 2022-08-08:
Thanks @eguidieri 

Which commit are you using for Gapuino RevA?