GreenWaves-Technologies/gap_sdk #326: make all

ChenMathilda, 2022-05-27:


closed 2022-06-01 by ChenMathilda

ChenMathilda, 2022-06-01:
> Hi， I tried `make clean all platform=gvsoc` under `/pulp-dronet/pulp-dronet-v2/gapflow`. But met the following errors: https://pastebin.pl/view/9e1a8baf.
> 
> ```
> # part of Error log
> Error: Graph stacked tensor S16_Output, input tensor Output_1 is also defined as a graph local or graph argument
> Execution aborted
> ```
> 
> I don't konw how to solve it. Hoping I can get solutions here.
> 
> My code environment: ` gap_sdk = 3.8, python =3.7, ubuntu18.04`. Using the onnx export by` /pulp-dronet/pulp-dronet-v2/model/dronet_v2_gapflow_original_himax.pth` and ` /pulp-dronet/pulp-dronet-v2/model/dronet_v2_gapflow_original.pth` I tried `gap_sdk = 3.9` but got more Errors: https://pastebin.pl/view/8518979a
> 
> Regards, Mathilda

solution: gap_sdk = 3.9

sousoux, 2022-06-01:
Is this solved? If not please post a link to the ONNX graph.

ChenMathilda, 2022-06-02:
> Is this solved? If not please reopen and post a link to the ONNX graph.

Yes, it solved. it was because the gap_sdk's version. gap_sdk >= v3.9.1