GreenWaves-Technologies/gap_sdk #328: SPI sending method and SD library

ROMAINPC, 2022-06-20:
Hello,

I'm trying to write on an SD card from GAPPoc-A board.
Before using a filesystem I need to implement an SD driver, there are many on the web.

The SPI of the `spi.h` library has certain advantages, in particular the automatic management of the CS.
But there is a big issue, the data buffer to read or write must be global. If I don't, the values sent are random.
According to the documentation it also must be aligned and a multiple of 4 bytes. 

Most SD libraries regularly perform small transactions. They use internal variables, macros or passing arguments by value.
It therefore seems very complicated to me to be able to achieve this functionality on GAP8. Has anyone managed to write to SD card? Is there another way to use the SPI channel?

Best Regards,
Romain.

Yaooooo, 2022-06-20:
Hi,

The buffer is not have to be global, BUT need to be placed in L2 memory. 
Since in GAP8, FC has its small L1 memory, where the stack is, so your buffer defined in the stack will be in L1 of FC. That's why you got random value. 
To reslove this, 2 options:
- Use dynamic buffer with malloc: manage your buffer by using `pi_l2_malloc` and `pi_l2_free`
- Use predefined L2 memory attribute `PI_L2` when you define your buffer.

Regards,
Yao

ROMAINPC, 2022-06-21:
Hi,
Thank you for your quick reply. The use of `PI_L2` is good to know.
I am implementing an sd driver, I try with the [lib from spidru user](https://github.com/spidru/sdmmc)  where I just need to glue the send and receive functions :
```C
int8_t spi_read(uint8_t *data, uint16_t data_size_bytes) {
    pi_spi_receive(&spi_dev, data, 8 * data_size_bytes, 0);
    return 0;
}
```
So I'm currently reworking the code to use intermediate buffer for each transaction,. To be sure I declared all my buffers like this, even if only index 0 is used : `static PI_L2 uint8_t cmd_buffer[4] __attribute__((aligned(4)));`

Then here is an example to send a single word to SD card : 
```C
uint8_t cmd = SDMMC_CMD0; // =0x40
cmd_buffer[0] = cmd;
dev->write(cmd_buffer, 1);
```
I keep informed in case of successful implementation of the driver.
Regards,
Romain

ROMAINPC, 2022-06-21:
Hello again,

I can send commands correctly to the sd card reader .

However the card never answer on the MISO pin.
I forced the configuration of the CS pin as an output, then with `RT_SPIM_CS_NONE` I can use `pi_gpio_pin_write()` to manage manually CS. I'm so sure that the sd card receive the correct CS.

By comparing the outputs with an arduino card also in CPOL=0 and CPHA=0, I think that the problem comes from the MOSI which is reversed.  SD protocol need 80 dummy cycles with CS high and MOSI high
**Is that possible to configure the MOSI pin to send 1 as low and keep high otherwise ?** I tried to configure the mosi gpio but it breaks the SPI send.

Regards,
Romain

ROMAINPC, 2022-07-05:
HI,
I did more tests. I had some issue with bad contacts, insufficient power or SD card that does not read,...

I finally managed to get multiple SD cards to respond with multiple readers. But I always end up with the same problem.
(Same work with CPOL=CPHA=0 or 1)
- `CMD0` is OK
- SD card answers `0x01` meaning command is valid
- `CMD8` is OK (and legal on all my SD cards, so i have only V2 cards)
- But... the board answer `0x01` but not at time ! The first bit is send at the same time of the last MOSI bit. So when my lib try to read command answer, it get `0x03`. (It is the `0x01` but with addition high value at right)

I would really like to be sure that the problem is not with the reversed mosi.

Additional resources about SD SPI protocol : [here](http://www.dejazzer.com/ee379/lecture_notes/lec12_sd_card.pdf) and [here](http://elm-chan.org/docs/mmc/mmc_e.html)

ROMAINPC, 2022-07-08:
Hello,
I solved my problem. :heavy_check_mark: 
There was indeed a problem because of the state of the MOSI. Failing to invert its behavior so that it remains in the high state during the SPI reads, it suffices to transmit `0xFF` in parallel with the reading.
So here is the read function implemented for spidru library : 
```C
static PI_L2 uint8_t high[8] __attribute__((aligned(32))) = {
    0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff};
int8_t spi_read(uint8_t *data, uint16_t data_size_bytes) {
    pi_spi_transfer(&spi_dev, high, data, 8 * data_size_bytes,
                    RT_SPIM_CS_NONE | PI_SPI_LINES_SINGLE);
    return 0;
}
```

I mark the issue solved. It remains to fix several problems with the library of spidru and fatFS. I'll keep it up if I have time.

closed 2022-07-08 by ROMAINPC