GreenWaves-Technologies/gap_sdk #332: Stuck at "load image Done" when uploading blink_led example

SamuelMathieuGPHY, 2022-06-30:
Hello, 

I'm currently trying to upload the example Blink_led to a GAPPoc-B.  When using the command line : 
make clean all run PMSIS_OS=freertos platform=board

The terminal return : 
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
TARGET create
GAP8 INIT TARGET
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
--------------------------
begining flash session (hyperflash)
--------------------------
load flasher to L2 memory
Loading binary through JTAG
[Flahser]: Hyperflash flasher is ready
Instruct flasher to begin flash per se
device struct address is 470047192
going to wait on addr GAP_RDY
wait on gap_rdy done witg buff ptr 0 469784844
loading image with addr 469784844 addr_min 469784844 and size 46112
load image done

It's then stuck on "load image done" forever. The only way to stop it is to unconnect the programmer.

My guess is that the SDK can write the program to the GAP8, but can't read it to make sure it's been properly written.

Ubuntu 20.04 is installed on a virtual machine using Oracle VM VirtualBox. I'm using a j-link edu programmer (https://www.segger.com/products/debug-probes/j-link/models/j-link-edu/) with a Cortex SWD adapter (https://www.adafruit.com/product/2094) to program the GAPPoc-B through JTAG. J-link as been installed on the VM.

The programmer as been set using the command line :
export GAPY_OPENOCD_CABLE=interface/jlink.cfg

suadsuljic7, 2022-10-31:
Hi, 
Did you manage to flash the board? 
I'm having the same issue same j-link and arm jtag 20-10 adapter with the sdk (release-v4.22.0). With the release-v4.0.0, I'm able to flash the board but I need the newest sdk version.

SamuelMathieuGPHY, 2022-10-31:
Hi,
I don't think we found the exact solution. I do remember that we started over on a new Virtual Machine and change something to the sdk when installing to make it work.
I don't know your situation, but I suggest you use a Olimex programmer. From my experience, it's a lot easier to work with the sdk and about 2 as fast when programming (make a huge difference when programming the occupancy management code).