GreenWaves-Technologies/gap_sdk #335: Using UART 1 in GAP9 on GVSOC

Irdroid, 2022-07-13:
Hi, 

Is it possible to use the UART1 interface in GVSOC ?
If yes, what configuration is needed in properties.json , uart_tb.json and in padframe.json?

Thanks in advance!

Yaooooo, 2022-07-13:
For GAP9, please report in gap9 sdk repository, thanks

closed 2022-07-13 by Yaooooo

Irdroid, 2022-07-14:
But Gap9 is target inside this SDK, no?
If not, could you provide a link to the GAP9 SDK in Github, 

thanks

fernandoFernandeSantos, 2022-11-28:
Hello

@Irdroid, were you able to download the SDK for GAP9? 
If not, @Yaooooo, where can I find the SDK that supports GAP9 EVK v2.0?

Thanks

Yaooooo, 2022-11-28:
Hi,

GAP9 is still under NDA, if you want to have GAP9 info and SDK, please contact our sales team by https://greenwaves-technologies.com/contacts/
Or directly send an email to sales@greenwaves-technologies.com

Thanks