GreenWaves-Technologies/gap_sdk #346: gapuino V1 error message

fernandoFernandeSantos, 2022-08-04:
Hello

I have a gapuino V1 and I'm using the SDK [f0a594c2a6272837e5b4aedc0aa4985952ab40ac](https://github.com/GreenWaves-Technologies/gap_sdk/commit/f0a594c2a6272837e5b4aedc0aa4985952ab40ac) commit.
However, when I try to build the SDK with the sourceme.sh setting to Gapuino V1, the following error happens:

```
if [ -e archi_pulp ]; then \
	cd archi_pulp && \
		export INSTALL_DIR=/home/fernando/gap8/gap_sdk/gap8/rtos/pulp/archi/workstation && \
		export TARGET_INSTALL_DIR=/home/fernando/gap8/gap_sdk/gap8/rtos/pulp/archi/target && \
		make build; \
fi
make[3]: Entering directory '/home/fernando/gap8/gap_sdk/gap8/rtos/pulp/archi_pulp'
scons
scons: Reading SConscript files ...
b'\x1b[91mFATAL ERROR: \x1b[0mExpecting property name enclosed in double quotes: line 14 column 7 (char 274)\n'
CalledProcessError: Command '['plpfiles', 'copy', '--item=archi_files']' returned non-zero exit status 255.:
  File "/home/fernando/gap8/gap_sdk/gap8/rtos/pulp/archi_pulp/SConstruct", line 21:
    for file in subprocess.check_output(shlex.split('plpfiles copy --item=archi_files')).decode('UTF-8').split():
  File "/usr/lib/python3.8/subprocess.py", line 415:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 516:
    raise CalledProcessError(retcode, process.args,
make[3]: *** [Makefile:2: build] Error 2
make[3]: Leaving directory '/home/fernando/gap8/gap_sdk/gap8/rtos/pulp/archi_pulp'
make[2]: *** [Makefile:7: do.archi] Error 2
make[2]: Leaving directory '/home/fernando/gap8/gap_sdk/gap8/rtos/pulp'
make[1]: *** [gap8/gap8.mk:108: pulp-os] Error 2
make[1]: Leaving directory '/home/fernando/gap8/gap_sdk'
make: *** [Makefile:42: all] Error 2
```

If I set the sourceme.sh to Gapuino V2, the SDK builds fine, but when I try to execute the helloworld example from the [GAP Manual](https://greenwaves-technologies.com/manuals/BUILD/HOME/html/index.html) the following error happens:

```
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
TARGET create
GAP8 INIT TARGET
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Error: timed out while waiting for target halted

Traceback (most recent call last):
  File "/home/fernando/gap8/gap_sdk/utils/gapy/gapy", line 162, in <module>
    main()
  File "/home/fernando/gap8/gap_sdk/utils/gapy/gapy", line 152, in main
    operationFunc(args, config, system)
  File "/home/fernando/gap8/gap_sdk/utils/gapy/run.py", line 189, in operationFunc
    raise RuntimeError('Runner has failed with value: %d' % status)
RuntimeError: Runner has failed with value: -1
make: *** [/home/fernando/gap8/gap_sdk/rtos/freeRTOS/vendors/gwt/rules/freeRTOS_rules.mk:417: flash_fs] Error 1
```

How can I solve this issue? I would appreciate any help thanks

eguidieri, 2022-08-08:
I had the same problem, answered in this other topic

https://github.com/GreenWaves-Technologies/gap_sdk/issues/320#issuecomment-1201316083

fernandoFernandeSantos, 2022-08-19:
Thanks @eguidieri however it does not work for the commit that I'm using, I don't know why.
Do you know if they abandoned the support for Gapuino V1 on the last releases of the SDK?