GreenWaves-Technologies/gap_sdk #347: L1_Memory can't be freed: examples/autotiler/BilinearResize

DYX-JL, 2022-08-05:
In examples/autotiler/BilinearResize, I tried to measure the remaining L1 Memory before and after executing the task of resizing image:

    printf("\nL1 remaining space size: % 8d bytes\n\n", rt_alloc_l1(0)->first_free->size);

    /* Allocating L1 memory for cluster */
    Resize_L1_Memory = (char *) pi_l1_malloc(&cluster_dev, _Resize_L1_Memory_SIZE);
    if (Resize_L1_Memory == 0)
    {
        printf("Failed to allocate %d bytes for L1_memory\n", _Resize_L1_Memory_SIZE);
        pmsis_exit(-5);
    }
    printf("\nL1 remaining space size: % 8d bytes\n\n", rt_alloc_l1(0)->first_free->size);

    ArgCluster_T cluster_call;

    //Assinging all input variables to Cluster structure
    cluster_call.ImageIn     = ImageIn;
    cluster_call.Win         = w_in;
    cluster_call.Hin         = h_in;
    cluster_call.Wout        = w_out;
    cluster_call.Hout        = h_out;
    cluster_call.ImageOut    = ImageOut;

    /* Prepare task to be offload to Cluster. */
    struct pi_cluster_task task = {0};
    task.entry = (void *) cluster_main;
    task.arg = &cluster_call;
    task.stack_size = (uint32_t) STACK_SIZE;

    /* Execute the function "cluster_main" on the Core 0 of cluster. */
    pi_cluster_send_task_to_cl(&cluster_dev, &task);

    pi_l1_free(&cluster_dev, Resize_L1_Memory, _Resize_L1_Memory_SIZE);
    printf("\nL1 remaining space size: % 8d bytes\n\n", rt_alloc_l1(0)->first_free->size);

And the result is:
```
L1 remaining space size:    65496 bytes

L1 remaining space size:    16544 bytes

L1 remaining space size:    48952 bytes
```
I can't understand why the `rt_alloc_l1(0)->first_free->size` returns 48952 instead of 65496 after I have freed the `Resize_L1_Memory` .
I think that the `rt_alloc_l1(0)->first_free->size` returns the remaing L1 Memory size, did I misunderstand it？

Pafrak, 2022-08-05:
Hello @DYX-JL,
 when you spawn a task also the stacks for each core are allocated and if you do not allocated/deallocate esplicitly they are kept allocated for next task. In the pi_cluster_task  you should be able to pass a pointer for the staks.

DYX-JL, 2022-08-06:
> Hello @DYX-JL, when you spawn a task also the stacks for each core are allocated and if you do not allocated/deallocate esplicitly they are kept allocated for next task. In the pi_cluster_task you should be able to pass a pointer for the staks.

Thanks for your reply,
If I want to allocate L1 Memory larger than 48952 bytes for another NN  after executing the task of ResizeImage, what shoud I do? Now, I failed to allocate L1 Memory of 50000 bytes for anthoer NN, beacuse only 48952 bytes can be used.

DYX-JL, 2022-08-16:
> Hello @DYX-JL, when you spawn a task also the stacks for each core are allocated and if you do not allocated/deallocate esplicitly they are kept allocated for next task. In the pi_cluster_task you should be able to pass a pointer for the staks.

Hi,
How could I explicitly free the L1 memory allocated by task (stack & slave stacks) ? Which function can I use?

closed 2022-10-19 by Doraemon-DYX