GreenWaves-Technologies/gap_sdk #348: "Total minimum memory requirement report" reporting negative memory

finnfi, 2022-08-08:
What does that mean if the required total memory is negative? 
[image]

DYX-JL, 2022-08-11:
Hi，
I got the same question, and I don't know the meaning of these six parameters, do you have any idea?

finnfi, 2022-08-15:
> 

No, unfortunately not.

Pafrak, 2022-08-16:
Hello @finnfi and @DYX-JL ,
 When a graph is generated it need to store in L2 and optionally in L3 the constants of a Graph, the weight and biases, and also the dynamic values, which are the activations (the temporary output of each of the layers). The negative value is surely a bug, is it possible to provide the code you use to generate that, so that we can debug it?

Thanks

finnfi, 2022-08-16:
More information can be found in #339. Is there anything more you need?

DYX-JL, 2022-08-19:
> More information can be found in #339. Is there anything more you need?
@Pafrak @finnfi 
Hi, 
Actually, my NN is also the depth estimation model, and I also got the wrong output on GAPUINO. However, I found that I can get correct output if I use `dump` command in nntool. I'm not sure that if this wrong output is caused by the intensive output feature map ( my output size is 224 x 224 x 1, which is the estimated depth map ).

finnfi, 2022-08-19:
@DYX-JL 
Yeah we did also produce right output with the dump command directly in nntool. Seems like there is a problem with the autotiler code. Do you quantize your graph with the nntool or do you do it before? And do you have any shared weights between layers?

DYX-JL, 2022-08-21:
> @DYX-JL Yeah we did also produce right output with the dump command directly in nntool. Seems like there is a problem with the autotiler code. Do you quantize your graph with the nntool or do you do it before? And do you have any shared weights between layers?

@finnfi 
Yeah, actually we do quantize the graph in nntool with the `aquant` commond. We have already deployed serveral NN models, this is the first time for us to encounter this problem.  We also guess that it seems like there is a problem with the autotiler code. Unfortunately, we don't know how to debug it.