GreenWaves-Technologies/gap_sdk #350: pi_fs_mount() fails

LorenzoLamberti94, 2022-09-06:
Hello,
I am facing a quite strange issue. 
I have a board with gap8 that is now failing to mount the FS.
Note: ive been using this board for a while and the FS mount was always successful.

This is the code i am running

```
    struct pi_device flash;
    struct pi_device fs;
    struct pi_readfs_conf conf0;

    // Open filesystem on flash
    pi_readfs_conf_init(&conf0);
    conf0.fs.flash = &flash;
    pi_open_from_conf(&fs, &conf0);
    if (pi_fs_mount(&fs))
    {
        printf("Error FS mounting ! \n");
        pmsis_exit(-2);
    }
```

Other notes:
I got this behavior after flashing the board with some new code, I am experimenting with setting Himax registers.
It is possible that i accidentally tried  to write to a wrong address using `pi_camera_reg_set()` function ?

Pafrak, 2022-09-08:
Ciao Lorenzo,
 do you open the flash before opening the fs?

LorenzoLamberti94, 2022-09-08:
Ciao francesco, 
Yes I do! here's the full code

```
    // Init Flash
    pi_hyperflash_conf_init(&flash_conf);
    pi_open_from_conf(&flash, &flash_conf);
    if (pi_flash_open(&flash))
    {
        printf("Error flash open ! \n");
        pmsis_exit(-3);
    }

    // Open filesystem on flash
    pi_readfs_conf_init(&conf0);
    conf0.fs.flash = &flash;
    pi_open_from_conf(&fs, &conf0);
    if (pi_fs_mount(&fs))
    {
        printf("Error FS mounting ! \n");
        pmsis_exit(-2);
    }
```