GreenWaves-Technologies/gap_sdk #352: Input error: The board returned an error: 256

prabhu-GW, 2022-10-08:
/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/gapy_v2/bin/gapy --target=gap9.gap9_v2 --target-dir=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/gapy_v2/targets --platform=board --work-dir=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/examples/gap9/basic/helloworld/BUILD/GAP9_V2/GCC_RISCV_FREERTOS --binary=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/examples/gap9/basic/helloworld/BUILD/GAP9_V2/GCC_RISCV_FREERTOS/test --config-opt=**/runner/boot/mode=jtag    --flash-property=0@flash:lfs:size --openocd-cable= --openocd-script=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/openocd_tools/tcl/gap9revb.tcl --openocd-tools=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/openocd_tools --flash-property=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/ssbl/bin/ssbl-gap9_evk@mram:rom:binary image
/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/gapy_v2/bin/gapy --target=gap9.gap9_v2 --target-dir=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/gapy_v2/targets --platform=board --work-dir=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/examples/gap9/basic/helloworld/BUILD/GAP9_V2/GCC_RISCV_FREERTOS --binary=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/examples/gap9/basic/helloworld/BUILD/GAP9_V2/GCC_RISCV_FREERTOS/test --config-opt=**/runner/boot/mode=jtag    --flash-property=0@flash:lfs:size --openocd-cable= --openocd-script=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/openocd_tools/tcl/gap9revb.tcl --openocd-tools=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/openocd_tools --flash-property=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/ssbl/bin/ssbl-gap9_evk@mram:rom:binary flash
/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/gapy_v2/bin/gapy --target=gap9.gap9_v2 --target-dir=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/gapy_v2/targets --platform=board --work-dir=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/examples/gap9/basic/helloworld/BUILD/GAP9_V2/GCC_RISCV_FREERTOS --binary=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/examples/gap9/basic/helloworld/BUILD/GAP9_V2/GCC_RISCV_FREERTOS/test --config-opt=**/runner/boot/mode=jtag    --flash-property=0@flash:lfs:size --openocd-cable= --openocd-script=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/openocd_tools/tcl/gap9revb.tcl --openocd-tools=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/openocd_tools --flash-property=/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/ssbl/bin/ssbl-gap9_evk@mram:rom:binary run
Launching execution with command:
openocd -d0 -c "gdb_port disabled; telnet_port disabled; tcl_port disabled" -f "" -f "/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/openocd_tools/tcl/gap9revb.tcl" -c "load_and_start_binary /home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/examples/gap9/basic/helloworld/BUILD/GAP9_V2/GCC_RISCV_FREERTOS/test 0x1c010100"
Open On-Chip Debugger 0.10.0+dev-00841-g1449af5bd (2022-10-08-16:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.htmlCan you
debug_level: 0

embedded:startup.tcl:26: Error: failed to load file "/home/sagar/gap9sdk_v2/Gap9SDK_Private/gap_sdk_private-master/utils/openocd_tools/": Is a directory
in procedure 'script' 
at file "embedded:startup.tcl", line 26
Input error: The board returned an error: 256

Can you please help why we are getting this error in Gap9 evalution board when flashing

CKRiedel, 2022-10-10:
I have the same problem, fresh SDK pull, try to run the example/gap9/basic/helloworld example

openocd -d0 -c "gdb_port disabled; telnet_port disabled; tcl_port disabled" -f "" -f "/home/gap9/Software/gap_sdk_private/utils/openocd_tools/tcl/gap9revb.tcl" -c "load_and_start_binary /home/gap9/Software/gap_sdk_private/examples/gap9/basic/helloworld/BUILD/GAP9_V2/GCC_RISCV_FREERTOS/test 0x1c010100"
Open On-Chip Debugger 0.10.0+dev-00841-g1449af5bd (2022-10-07-15:06)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0

embedded:startup.tcl:26: Error: failed to load file "/home/gap9/Software/gap_sdk_private/utils/openocd_tools/": Is a directory
in procedure 'script' 
at file "embedded:startup.tcl", line 26
Input error: The board returned an error: 256
make: *** [/home/gap9/Software/gap_sdk_private/rtos/pmsis/os/freeRTOS/vendors/gwt/rules/freeRTOS_rules.mk:473: run] Error 1

The file "/home/gap9/Software/gap_sdk_private/utils/openocd_tools/tcl/gap9revb.tcl" exists

ls -l /home/gap9/Software/gap_sdk_private/utils/openocd_tools/tcl/gap9revb.tcl
-rw-rw-r-- 1 gap9 gap9 899 Okt  7 14:25 /home/gap9/Software/gap_sdk_private/utils/openocd_tools/tcl/gap9revb.tcl

SebastianNemes, 2023-01-19:
Hello,
Is this issue solved? If so, what was the solution?