GreenWaves-Technologies/gap_sdk #356: FC to access the CL L1 memory

fernandoFernandeSantos, 2022-10-28:
Hello

Is it possible for the FC core to access the cluster L1 memory? I have tried the following:
```c
...
struct pi_device cluster_dev;
struct pi_cluster_conf cl_conf;
pi_cluster_conf_init(&cl_conf);
cl_conf.id = 0;
pi_open_from_conf(&cluster_dev, &cl_conf);
int elements = (62 * 1024) / sizeof(int);
int MEMORY = (int *) pi_cl_l1_malloc(&cluster_dev, elements * sizeof(int));
for (uint32_t i = 0; i < elements; i++) { // runs on the FC core
        MEMORY[i] = default_val;
}
...
```
The code runs ok on GAP8v2. However, I don't know if I'm, in fact, setting values on the cluster L1 memory. 

For the moment, I'm not worried about the performance when using cluster L1 memory from the FC core, I just need to know if it is possible or not.

Yaooooo, 2022-10-28:
Hello,

Yes you can and it should do the job. 
BUUUUUUUUUT!!!! We stronly recommand NOT use it like this, because:
* Access L1 from FC or Access L2 from Cluster cores are OK, BUT around 10 times slower. 

Solution is easy, you can do it from the cluster side. So you can just put this for loop into the cluster task.

Furthermore, if it's a big buffer, to optimize it, you can use 8 cores to do that.

fernandoFernandeSantos, 2022-10-28:
Thanks @Yaooooo
I'll use the cluster to access the L1 in the implementation. I just needed to make sure that this approach also works

closed 2022-10-28 by fernandoFernandeSantos