GreenWaves-Technologies/gap_sdk #361: Is it possible for enabling multi-thread inference on GAP8

huzq85, 2022-12-05:
hi,
Recently I have been investigating whether we could enable the inference job by multi-threadings. I went through some examples (MoboileNet model from nn_menu), and found that all the inference tasks are single-threaded. Meanwhile, it looks like GAP8 platform could perform multi-threading tasks from the Helloworld example (`examples/gap8/basic/helloworld`).
As such, I am wondering if we could enable the multi-thread tasks when we are doing DNN inference.

sousoux, 2022-12-07:
all kernels on gap are parallel. (the threads are cores and they are scheduled across all of them)

closed 2022-12-07 by sousoux

huzq85, 2022-12-24:
> all kernels on gap are parallel. (the threads are cores and they are scheduled across all of them)

Got you. Thanks!