GreenWaves-Technologies/gap_sdk #362: Profiler: power not working in GVSOC settings

m1karmida, 2022-12-22:
Hi, i would like to show 'power' values in the Profiler, i saw it compares as choice in GVSOC settings but it's not possible enable them. This is because of running on gvsoc or are there other issues? Thank you.

Yaooooo, 2022-12-26:
Hi, not sure that I understand what you would like to do. 
'power' values, you mean you want to check the power of a number? 
Could you please provide more details, or a piece of code to show what you would like to do, please?

m1karmida, 2022-12-26:
[image]
These are the possible choices in the gvsoc settings. I would like to show power details (voltage, power consuption,etc) profiling the application. The 'power' choice can't be enabled clicking on it.

Yaooooo, 2022-12-26:
OK, understood.
Unfortunately, the power unit hasn't been implemented yet, so we cannot see
the power simulation in GVSOC.
If you want to understand the power side, it's better to measure on the
chip.

m1karmida, 2022-12-26:
Thank you for the fast response.

closed 2022-12-26 by m1karmida