GreenWaves-Technologies/gap_sdk #365: OPENOCD issue when install GAP8_SDK

aigrison, 2023-01-06:
Hello,
I use UBUNTU 22.04.01 LTS 
I tried to install Gap8_SDK but i have an issue with openocd.
After Autotiler et license agrement i had this error 

Makefile:5643: avertissement : surchargement de la recette pour la cible « check-recursive »
Makefile:4413: avertissement : ancienne recette ignorée pour la cible « check-recursive »
make  all-recursive
make[1] : on entre dans le répertoire « /home/user/gap8_openocd »
Makefile:5643: avertissement : surchargement de la recette pour la cible « check-recursive »
Makefile:4413: avertissement : ancienne recette ignorée pour la cible « check-recursive »
Making all in jimtcl
make[2] : on entre dans le répertoire « /home/user/gap8_openocd/jimtcl »
make[2]: rien à faire pour « all ».
make[2] : on quitte le répertoire « /home/user/gap8_openocd/jimtcl »
Making all in src/jtag/drivers/libjaylink
make[2] : on entre dans le répertoire « /home/user/gap8_openocd/src/jtag/drivers/libjaylink »
make  all-recursive
make[3] : on entre dans le répertoire « /home/user/gap8_openocd/src/jtag/drivers/libjaylink »
Making all in libjaylink
make[4] : on entre dans le répertoire « /home/user/gap8_openocd/src/jtag/drivers/libjaylink/libjaylink »
make[4]: rien à faire pour « all ».
make[4] : on quitte le répertoire « /home/user/gap8_openocd/src/jtag/drivers/libjaylink/libjaylink »
make[4] : on entre dans le répertoire « /home/user/gap8_openocd/src/jtag/drivers/libjaylink »
make[4] : on quitte le répertoire « /home/user/gap8_openocd/src/jtag/drivers/libjaylink »
make[3] : on quitte le répertoire « /home/user/gap8_openocd/src/jtag/drivers/libjaylink »
make[2] : on quitte le répertoire « /home/user/gap8_openocd/src/jtag/drivers/libjaylink »
make[2] : on entre dans le répertoire « /home/user/gap8_openocd »
Makefile:5643: avertissement : surchargement de la recette pour la cible « check-recursive »
Makefile:4413: avertissement : ancienne recette ignorée pour la cible « check-recursive »
depbase=`echo src/flash/nor/numicro.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/bash ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/share/gap8-openocd/openocd\" -DBINDIR=\"/usr/bin\" -I./jimtcl -I./jimtcl  -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Werror -g -O2 -MT src/flash/nor/numicro.lo -MD -MP -MF $depbase.Tpo -c -o src/flash/nor/numicro.lo src/flash/nor/numicro.c &&\
mv -f $depbase.Tpo $depbase.Plo
depbase=`echo src/target/arm920t.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/bash ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/share/gap8-openocd/openocd\" -DBINDIR=\"/usr/bin\" -I./jimtcl -I./jimtcl  -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Werror -g -O2 -MT src/target/arm920t.lo -MD -MP -MF $depbase.Tpo -c -o src/target/arm920t.lo src/target/arm920t.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/share/gap8-openocd/openocd\" -DBINDIR=\"/usr/bin\" -I./jimtcl -I./jimtcl -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Werror -g -O2 -MT src/target/arm920t.lo -MD -MP -MF src/target/.deps/arm920t.Tpo -c src/target/arm920t.c -o src/target/arm920t.o
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/share/gap8-openocd/openocd\" -DBINDIR=\"/usr/bin\" -I./jimtcl -I./jimtcl -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Werror -g -O2 -MT src/flash/nor/numicro.lo -MD -MP -MF src/flash/nor/.deps/numicro.Tpo -c src/flash/nor/numicro.c -o src/flash/nor/numicro.o
src/flash/nor/numicro.c: In function ‘numicro_fmc_cmd’:
src/flash/nor/numicro.c:1244:17: error: this ‘if’ clause does not guard... [-Werror=misleading-indentation]
 1244 |                 if (retval != ERROR_OK)
      |                 ^~
In file included from ./src/flash/common.h:21,
                 from src/flash/nor/core.h:25,
                 from src/flash/nor/imp.h:22,
                 from src/flash/nor/numicro.c:32:
./src/helper/log.h:119:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  119 |         do { \
      |         ^~
src/flash/nor/numicro.c:1246:25: note: in expansion of macro ‘LOG_DEBUG’
 1246 |                         LOG_DEBUG("status: 0x%" PRIx32 "", status);
      |                         ^~~~~~~~~
src/flash/nor/numicro.c: In function ‘numicro_erase’:
src/flash/nor/numicro.c:1512:25: error: this ‘if’ clause does not guard... [-Werror=misleading-indentation]
 1512 |                         if (retval != ERROR_OK)
      |                         ^~
In file included from ./src/flash/common.h:21,
                 from src/flash/nor/core.h:25,
                 from src/flash/nor/imp.h:22,
                 from src/flash/nor/numicro.c:32:
./src/helper/log.h:119:9: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  119 |         do { \
      |         ^~
src/flash/nor/numicro.c:1514:33: note: in expansion of macro ‘LOG_DEBUG’
 1514 |                                 LOG_DEBUG("status: 0x%" PRIx32 "", status);
      |                                 ^~~~~~~~~
src/flash/nor/numicro.c: In function ‘numicro_write’:
src/flash/nor/numicro.c:1601:33: error: this ‘if’ clause does not guard... [-Werror=misleading-indentation]
 1601 |                                 if (retval != ERROR_OK)
      |                                 ^~
[... 33 lines cut ...]
src/target/arm920t.c:304:9: error: ‘arm9tdmi_write_core_regs’ accessing 64 bytes in a region of size 8 [-Werror=stringop-overflow=]
  304 |         arm9tdmi_write_core_regs(target, 0x3, regs);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/target/arm920t.c:304:9: note: referencing argument 3 of type ‘uint32_t *’ {aka ‘unsigned int *’}
In file included from src/target/arm920t.h:22,
                 from src/target/arm920t.c:24:
src/target/arm9tdmi.h:40:6: note: in a call to function ‘arm9tdmi_write_core_regs’
   40 | void arm9tdmi_write_core_regs(struct target *target,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:3466 : src/target/arm920t.lo] Erreur 1
make[2] : on quitte le répertoire « /home/user/gap8_openocd »
make[1]: *** [Makefile:4413 : all-recursive] Erreur 1
make[1] : on quitte le répertoire « /home/user/gap8_openocd »
make: *** [Makefile:2061 : all] Erreur 2
[log.txt](https://github.com/GreenWaves-Technologies/gap_sdk/files/10361499/log.txt)

My PC is a DELL G15 5520 :
Processeur Intel® Core™ i7-12700H de 12e gen 
graphic card NVIDIA® GeForce RTX™ 3060 avec 6 Go de mémoire GDDR616 Go,
2 x 8 Go de mémoire DDR5 bicanale à 4 800 MHz
Disque SSD M.2 PCIe NVMe de 512 Go
Dual boot UBUNTU 22.04.1 LTS & Win 11 

I m a noob so thx for helping

Yaooooo, 2023-01-09:
Hi,

Sorry, our SDK for GAP8 support only Ubuntu 18.04/20.04, and we don't have
resources to move to 22.04 for now.
If you'd like to stay on 22.04, you could try to compile the openOCD
manually, the source code is in
https://github.com/GreenWaves-Technologies/gap8_openocd
Please feel free to submit a pull request if you have fixed them.

Regards,

aigrison, 2023-01-12:
Hi, 
I changed my Ubuntu version, now on 20.04.5LTS
I have no issue during SDK install but when i try the example Helloworld i have this error code :

DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
TARGET create
GAP8 INIT TARGET
Error: libusb_claim_interface() failed with LIBUSB_ERROR_BUSY
Error: unable to open ftdi device with vid 15ba, pid 002b, description 'Olimex OpenOCD JTAG ARM-USB-OCD-H', serial '*' at bus location '*'

Traceback (most recent call last):
  File "/home/user/gap_sdk/utils/gapy/gapy", line 162, in <module>
    main()
  File "/home/user/gap_sdk/utils/gapy/gapy", line 152, in main
    operationFunc(args, config, system)
  File "/home/user/gap_sdk/utils/gapy/run.py", line 189, in operationFunc
    raise RuntimeError('Runner has failed with value: %d' % status)
RuntimeError: Runner has failed with value: -1
make: *** [/home/user/gap_sdk/utils/rules/pulp_rules.mk:246 : flash_fs] Erreur 1

I use GAPocB_2.4 board
ARM-USB-OCD-H rev E & ARM-JTAG-20-10 rev C
I have no error when i try with gvsoc platform.

Thanks

Yaooooo, 2023-01-12:
Seems a problem of USB HW or USB lib.
2 things to check:

1. Could you run *lsusb* to check if you can see the Olimex in the result?
with the same vid and pid as in the log?

   - If it's not in the list, please check the HW connection. If you are
   using a VM, please check if the USB is connected to your VM but not the host
   - if it's in the list, please follow the 2. below

2. Please follow the README doc, this section:
[image: image.png]

aigrison, 2023-01-16:
Hi, thx for your fast answer.
with lsusb : "Bus 003 Device 007: ID 15ba:002b Olimex Ltd. ARM-USB-OCD-H JTAG+RS232"
The vid and pid are matching with logs.
At the first try (unplug/plug USB) I have the message below : 
"Open On-Chip Debugger 0.10.0+dev-gb84d97ec4 (2023-01-12-13:44)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
TARGET create
GAP8 INIT TARGET
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: gap8.cpu: IR capture error; saw 0x00 not 0x01
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Error: Burst read failed
Error: Burst read failed
Error: Burst read failed
can't read halt state
Error: Couldn't read the CPU state
Error: Burst read failed
can't read halt state2
Init jtag
Initialising GAP8 JTAG TAP
Error: Burst read failed"
The process still endless so i kill it and when i make another try (without USB deconnection) i have the message i already gave you.
Can you provide me a link for the README doc ? i didn't find it...

thx

aigrison, 2023-01-16:
Ok I just understand... I don't kill the task when i closed the terminal. The task continue in background so when i try to launch another time the USB port is already busy!
The source issue is : 
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: gap8.cpu: IR capture error; saw 0x00 not 0x01
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Error: Burst read failed