GreenWaves-Technologies/gap_sdk #367: Error: Custom network for Ai-deck

godhj93, 2023-01-27:
Hi, all

I am trying to deploy custom neural networks to Ai-deck(crazyflie) and familiar with tensorflow and tensorflow lite but not NNtools.

I succeed the classification example for ai-deck, but When I try to my custom network, an error occurs.

one of my custom network's architecture is as follows:
```
class Encoder_tf(tf.keras.Model):
    
    def __init__(self, obs_shape):
        super().__init__()
        
        assert len(obs_shape) == 3
        
        self.repr_dim = 32 * 35 * 35 +(7*3)
        
        self.convnet = tf.keras.Sequential([
            tf.keras.layers.Conv2D(filters = 32, kernel_size = 3, strides = 2, activation='relu'),
            tf.keras.layers.Conv2D(filters = 32, kernel_size = 3, strides = 1, activation='relu'),
            tf.keras.layers.Conv2D(filters = 32, kernel_size = 3, strides = 1, activation='relu'),
            tf.keras.layers.Conv2D(filters = 32, kernel_size = 3, strides = 1, activation='relu'),
        ])     
        
        self.flatten = tf.keras.layers.Flatten()
        self.concat = tf.keras.layers.Concatenate()
        
    def call(self, x):
        #Input: 64x64x4
        (obs, drone_states) = x
        
        obs = obs / 255.0 - 0.5
        
        h = self.convnet(obs)
        h = self.flatten(h)
        
        drone_states = self.flatten(drone_states) # 9*9*32 = 2592
        print(h.shape)
        print(drone_states.shape)
        h = self.concat([h, drone_states])
        return h
       
```

and I run the command 

```
docker run --rm -v ${PWD}:/module aideck-with-autotiler tools/build/make-example examples/ai/classification clean model build image
```

the error  is too long so I attached a link to see it
(the link: https://godhj93.notion.site/github-issue-error-e39347a8562b4395a2a5e31bf9097948)

So far, I checked the concatenation layer and flatten layer are problem. when those layers are removed the command works well.

Is there any solution for my problem? 

Thanks

sousoux, 2023-01-27:
The problems are all very clearly indicated in compiler errors. The network seems to convert fine. I would suggest that you work on correcting the example code that you have clearly modified.

**/module/examples/ai/classification/BUILD_MODEL_SQ8BIT/classificationKernels.h:52:12: note: expected 'signed char * restrict' but argument is of type 'unsigned char *'**

classification.c:67:18: warning: passing argument 2 of 'classificationCNN' from incompatible pointer type [-Wincompatible-pointer-types]

**/module/examples/ai/classification/BUILD_MODEL_SQ8BIT/classificationKernels.h:52:12: note: expected 'signed char * restrict' but argument is of type 'short int *'**

classification.h:32:21: error: too few arguments to function 'classificationCNN'

classification.c:201:24: warning: passing argument 1 of 'pi_cluster_conf_init' from incompatible pointer type [-Wincompatible-pointer-types]

/gap_sdk/rtos/pmsis/pmsis_api/include/pmsis/cluster/cluster_sync/fc_to_cl_delegate.h:74:6: note: expected 'struct pi_cluster_conf *' but argument is of type 'struct pi_uart_conf *'

closed 2023-01-27 by sousoux

godhj93, 2023-01-30:
Hi, thank you for quick reaply.

We have not touched the code. we have just replaced the network file.

As mentioned above, when we removed concat and flatten layers the whole procedure completed just fine.

I want to know what differencies are when those layers exsits or not.

(Just to double check I have jut downloaded the examples again and run the same test with the same end result.)

Can you reopen this issue? 

thanks