GreenWaves-Technologies/gap_sdk #370: NNTOOL: Error when 'open  *.onnx'

ChenMathilda, 2023-02-12:
Hi, I tried to use `open `command in nntool to open the onnx model generated by the parameters trained by pulp-dronet-v2, but I encountered the following error:
```
(NNT) open model_gapflow.onnx 
open - opening graph file model_gapflow.onnx load_quantization = False
onnx - unable to determine batch dimension. if the graph fails to import properly set it to 1 or a variable.
Traceback (most recent call last):
  File "/home/chennaiting/anaconda3/envs/gap_sdk3.9/lib/python3.8/site-packages/cmd2/cmd2.py", line 1661, in onecmd_plus_hooks
    stop = self.onecmd(statement, add_to_history=add_to_history)
  File "/home/chennaiting/anaconda3/envs/gap_sdk3.9/lib/python3.8/site-packages/cmd2/cmd2.py", line 2081, in onecmd
    stop = func(statement)
  File "/home/chennaiting/anaconda3/envs/gap_sdk3.9/lib/python3.8/site-packages/cmd2/decorators.py", line 223, in cmd_wrapper
    return func(cmd2_app, args)
  File "/home/chennaiting/pulp-dronet/gap_sdk/tools/nntool/interpreter/commands/open.py", line 118, in do_open
    self.__open_graph(args)
  File "/home/chennaiting/pulp-dronet/gap_sdk/tools/nntool/interpreter/commands/open.py", line 92, in __open_graph
    G = create_graph(graph_file, opts=opts)
  File "/home/chennaiting/pulp-dronet/gap_sdk/tools/nntool/importer/importer.py", line 52, in create_graph
    graph = importer.create_graph(filename, opts)
  File "/home/chennaiting/pulp-dronet/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 54, in create_graph
    G = self._import_onnx_model(G, model.graph, opset_import, opts)
  File "/home/chennaiting/pulp-dronet/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 68, in _import_onnx_model
    self._import_nodes(G, graph, handlers, all_nodes, outputs, opts)
  File "/home/chennaiting/pulp-dronet/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 169, in _import_nodes
    params = handler.handle(OnnxNode(node), all_nodes=all_nodes, vars_dict=vars_dict,
  File "/home/chennaiting/pulp-dronet/gap_sdk/tools/nntool/importer/onnx/handlers/handler.py", line 63, in handle
    raise BackendIsNotSupposedToImplementIt(
onnx.backend.test.runner.BackendIsNotSupposedToImplementIt: Constant version 13 is not implemented.
EXCEPTION of type 'BackendIsNotSupposedToImplementIt' occurred with message: 'Constant version 13 is not implemented.'
```
Here is the network [dronet_v2_gapflow](https://github.com/pulp-platform/pulp-dronet/blob/master/pulp-dronet-v2/model/dronet_v2_gapflow.py)
Here is the .pth file [pulp_dronet_v2_10.pth](https://github.com/ChenMathilda/temp/blob/main/pulp_dronet_v2_10.pth)
Here is the .onnx file [model_gapflow.onnx](https://github.com/ChenMathilda/temp/blob/main/model_gapflow.onnx)

And my environment are lisred:
python: 3.8
gap_sdk: 3.9.1

I was able to run the `open` command successfully before, but I don't know if I made some changes to the gap_sdk environment that caused this problem.
In addition, I noticed that there is a `bn module `in .pth file, but there is no such module in the generated onnx model.This is different from the onnx model in the official [pulp-dronet-v2](https://github.com/pulp-platform/pulp-dronet/tree/master/pulp-dronet-v2) is as follow: :[model_original.onnx](https://github.com/pulp-platform/pulp-dronet/blob/master/pulp-dronet-v2/gapflow/nntool_input/models_onnx/model_original.onnx)
[image]

This question has been bothering me for a long time. I look forward to your reply!

sousoux, 2023-02-12:
Your SDK is very old. Please update it. ONNX Constant operator is supported in opset version 13 in the latest SDK.

(https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/nntool/nntool/importer/onnx/handlers/backend/constant.py)

closed 2023-02-12 by sousoux

ChenMathilda, 2023-02-16:
> ``> Your SDK is very old. Please update it. ONNX Constant operator is supported in opset version 13 in the latest SDK.
> 
> > (https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/nntool/nntool/importer/onnx/handlers/backend/constant.py)
> 
> Hi @sousoux, I changed gap_sdk version into 4.22.0, and can run 'open *.onnx' successfully! But when I tried to run`make clean all run platform=gvsoc` under path `/pulp-dronet-v2/gapflow`, I met this error:
> 
> ```
> (gap_sdk)
> # chennaiting @ seu-netsi in ~/pulp-dronet/pulp-dronet-v2/gapflow on git:main x [20:05:39]
> $ make clean all run platform=gvsoc
> APP_SRCS... main.c BUILD_MODEL_SQ8BIT/networkKernels.c /home/chennaiting/pulp-dronet/gap_sdk/libs/gap_lib/img_io/ImgIO.c /home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_Activation_SQ8.c /home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_Bias_Linear_SQ8.c /home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_Conv_SQ8.c /home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_Pooling_SQ8.c /home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_Conv_DW_SQ8.c /home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_MatAlgebra_SQ8.c /home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_SoftMax_SQ8.c /home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_AT_Misc.c /home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/RNN_SQ8.c /home/chennaiting/pulp-dronet/gap_sdk/tools/nntool/autotiler/kernels/norm_transpose.c /home/chennaiting/pulp-dronet/gap_sdk/tools/nntool/autotiler/kernels/copy.c
> APP_CFLAGS... -g -O3 -w -mno-memcpy -fno-tree-loop-distribute-patterns -I. -I/home/chennaiting/pulp-dronet/gap_sdk/libs/gap_lib/include -I/home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/Emulation -I/home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/Autotiler -I/home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries -I/home/chennaiting/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8 -I/home/chennaiting/pulp-dronet/gap_sdk/tools/nntool/autotiler/kernels -IBUILD_MODEL_SQ8BIT -DAT_MODEL_PREFIX=network -DAT_INPUT_HEIGHT=200 -DAT_INPUT_WIDTH=200 -DAT_INPUT_COLORS=1 -DSTACK_SIZE=6096 -DSLAVE_STACK_SIZE=1024 -DAT_IMAGE=/home/chennaiting/pulp-dronet/pulp-dronet-v2/gapflow/images/frame_2.pgm -DPERF -DMODEL_ID= -DFREQ_FC=250 -DFREQ_CL=175 -DAT_CONSTRUCT=networkCNN_Construct -DAT_DESTRUCT=networkCNN_Destruct -DAT_CNN=networkCNN -DAT_L3_ADDR=network_L3_Flash
> sudo rm -f BUILD_MODEL_SQ8BIT/GenTile
> sudo rm -f -rf BUILD_MODEL_SQ8BIT
> sudo rm -f -rf nntool_output
> **make: *** No rule to make target '/home/chennaiting/pulp-dronet/gap_sdk/tools/nntool/autotiler/generators/nntool_extra_generators.c', needed by 'BUILD_MODEL_SQ8BIT/GenTile'.  Stop.**
> ```
> 
> According to the error report, I deleted 55 lines of code in `pulp-dronet-v2/gapflow/common/model_decl.mk`
> 
> [image] It do make some progresses, but still encountered error:
> 
> ```
> # chennaiting @ seu-netsi in ~/pulp/pulp-dronet/pulp-dronet-v2/gapflow on git:master x [19:08:43]
> make clean all run platform=gvsoc
> APP_SRCS... main.c BUILD_MODEL_SQ8BIT/networkKernels.c /home/chennaiting/pulp/pulp-dronet/gap_sdk/libs/gap_lib/img_io/ImgIO.c /home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_Activation_SQ8.c /home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_Bias_Linear_SQ8.c /home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_Conv_SQ8.c /home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_Pooling_SQ8.c /home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_Conv_DW_SQ8.c /home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_MatAlgebra_SQ8.c /home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_SoftMax_SQ8.c /home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/CNN_AT_Misc.c /home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8/RNN_SQ8.c /home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/nntool//autotiler/kernels/norm_transpose.c /home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/nntool//autotiler/kernels/copy.c  
> APP_CFLAGS... -g -O3 -w -mno-memcpy -fno-tree-loop-distribute-patterns -I. -I/home/chennaiting/pulp/pulp-dronet/gap_sdk/libs/gap_lib/include -I/home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/Emulation -I/home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/Autotiler -I/home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries -I/home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8 -I/home/chennaiting/pulp/pulp-dronet/gap_sdk/tools/nntool//autotiler/kernels -IBUILD_MODEL_SQ8BIT -DAT_MODEL_PREFIX=network -DAT_INPUT_HEIGHT=200 -DAT_INPUT_WIDTH=200 -DAT_INPUT_COLORS=1 -DSTACK_SIZE=6096 -DSLAVE_STACK_SIZE=1024 -DAT_IMAGE=/home/chennaiting/pulp/pulp-dronet/pulp-dronet-v2/gapflow/images/frame_2.pgm -DPERF -DMODEL_ID= -DFREQ_FC=250 -DFREQ_CL=175 -DAT_CONSTRUCT=networkCNN_Construct -DAT_DESTRUCT=networkCNN_Destruct -DAT_CNN=networkCNN -DAT_L3_ADDR=network_L3_Flash
> rm -f BUILD_MODEL_SQ8BIT/GenTile
> rm -f -rf BUILD_MODEL_SQ8BIT
> rm -f -rf nntool_output 
> mkdir BUILD_MODEL_SQ8BIT
[... cut ...]
>    23 | #include "CNN_FloatType.h"
>       |          ^~~~~~~~~~~~~~~~~
> compilation terminated.
> make: *** [common/model_rules.mk:56: BUILD_MODEL_SQ8BIT/GenTile] Error 1
> ```
> 
> I think this problem is caused by the large span of changing gap_sdkv3.9 to gap_sdkv4.22 (some file location may already changed). I wonder if there are any solutions? Or I can just modify the pulp-dronet's makefile by my own?

I try to solve this problem by my own and do make some progresses.
If you have any suggestiones, pls tell me.