GreenWaves-Technologies/gap_sdk #373: GenTilingDebug cannot print the debugging message

huzq85, 2023-02-20:
hi,
Sorry to bother you right now but currently I meet an issue when I try to generate the tiled code.
The thing is that I want the debug information to be printed when generating the tiling code. For instance, the function `GenTilingDebug()` from the lib Autotiler has been called from the generator `/tools/autotiler_v3/CNN_Generators_SQ8/CNN_Generators_SQ8.c`. However, when generating the tiling code, the debug information cannot be printed. If I replace the function call `GenTilingDebug()` with `printf()`, then the information can be printed on the console (but I don't think it's a proper way to do so).
The comment for the function `GenTilingDebug()` says it's for `Generates an debug message. Could be deactivated`. The question is: how could I activate this function call?

Any input will be appreciated!

sousoux, 2023-02-20:
The compiled generator takes options. You can pass it -h to see a help message. The generator log level is set by --log_level=<[0,1,2]>. 2 will enable debug messages.

closed 2023-02-20 by sousoux

huzq85, 2023-03-21:
@sousoux got you, I will try it. Thanks!