GreenWaves-Technologies/gap_sdk #388: Unnable to setup the sdk, make doesn't work.

enzoniko, 2023-06-04:
Hello, 
I've got a GAP8 Gapuino board (with usb-b cable), I am trying to setup the board to test it with some machine learning algorithms.
It's been 1 week since I started trying, basically I've followed 3 different sets of instructions to setup the board, none of them worked.
The one I've got further was this one: https://greenwaves-technologies.com/manuals/BUILD/HOME/html/index.html.

I've followed every instruction at the best of my capabilities, I also made a 10 minutes video following them, the URL for the video that shows all the steps I did to try to install the GAP SDK is this one: https://youtu.be/Ru_d_Vaqff8. 

I am using a fresh Ubuntu 18.04 Bionic Beaver 64-Bit virtual machine just like in the instructions.

Basically all the instructions worked perfectly, up until the 'make minimal_sdk' part (the video shows this best, please take 10 min to watch it, I've edited it to make it quick). 

when I use the 'make minimal_sdk' command inside the gap_sdk directory this is logged: "make: *** No rule to make target 'minimal_sdk'.  Stop."

when I try something like 'make minimal' this is logged:

enzo@ubuntu:~/gap_sdk$ make minimal
[...]
Install file: "bin/plpinfo" as "/home/enzo/gap_sdk/install/workstation/bin/plpinfo"
Install file: "bin/plpfiles" as "/home/enzo/gap_sdk/install/workstation/bin/plpfiles"
make[3]: Leaving directory '/home/enzo/gap_sdk/utils/gap_configs'
if [ -e archi_pulp ]; then \
	cd archi_pulp && \
		export INSTALL_DIR=/home/enzo/gap_sdk/gap8/rtos/pulp/archi/workstation && \
		export TARGET_INSTALL_DIR=/home/enzo/gap_sdk/gap8/rtos/pulp/archi/target && \
		make build; \
fi
make[3]: Entering directory '/home/enzo/gap_sdk/gap8/rtos/pulp/archi_pulp'
scons
scons: Reading SConscript files ...
ImportError: No module named configparser:
  File "/home/enzo/gap_sdk/gap8/rtos/pulp/archi_pulp/SConstruct", line 4:
    import pulp_config as plpconfig
  File "/home/enzo/gap_sdk/utils/gap_configs/python/pulp_config.py", line 19:
    import json_tools as js
  File "/home/enzo/gap_sdk/utils/gapy/json_tools.py", line 23:
    import configparser
Makefile:2: recipe for target 'build' failed
make[3]: *** [build] Error 2
make[3]: Leaving directory '/home/enzo/gap_sdk/gap8/rtos/pulp/archi_pulp'
Makefile:7: recipe for target 'do.archi' failed
make[2]: *** [do.archi] Error 2
make[2]: Leaving directory '/home/enzo/gap_sdk/gap8/rtos/pulp'
gap8/gap8.mk:106: recipe for target 'pulp-os' failed
make[1]: *** [pulp-os] Error 2
make[1]: Leaving directory '/home/enzo/gap_sdk'
Makefile:42: recipe for target 'all' failed
make: *** [all] Error 2

However, when I run `python` and `import configparser` no error is shown, this means I do have a module named configparser.

Observation: In the video, I've followed two paths, installing the SDK inside and outside the toolchain directory, both failed.

Can someone please help me out, this is the first "job" i've got as an intern and I'd be really happy if it worked out.

sousoux, 2023-06-04:
Error from the text that you posted:

make[3]: Entering directory '/home/enzo/gap_sdk/gap8/rtos/pulp/archi_pulp'
scons
scons: Reading SConscript files ...
ImportError: No module named configparser:

Instructions from the readme that you "followed" :) :

Python requirements
Our modules (gapy runner) require a few additional Python packages that you can install with this command from GAP SDK root folder:

**pip3 install -r requirements.txt**
pip3 install -r doc/requirements.txt

Contents of requirements.txt:

pyelftools>=0.24
prettytable==0.7.2
pexpect>=4.0.0,<5
**configparser**
argcomplete
requests
six
kconfiglib

Please follow the install instructions precisely.

sousoux, 2023-06-04:
Almost certainly the SDK is not using the same python as you are on the commandline.

Put a:

import sys

print(sys.executable)

at the top of one of the scripts executed to find out which one.

e.g. /home/enzo/gap_sdk/utils/gap_configs/python/pulp_config.py

 Once you have figured out which python is being used make sure you have installed the requirements.

enzoniko, 2023-06-04:
Okay, I've put a `import sys` and a `print(sys.executable)` at the top of /home/enzo/gap_sdk/utils/gap_configs/python/pulp_config.py and I called the `make minimal` command in the gap_sdk directory.
This is the output:

make -f gap8/gap8.mk gap8.all
make[1]: Entering directory '/home/enzo/gap_sdk'
cp -rf /home/enzo/gap_sdk/utils/bin/* /home/enzo/gap_sdk/install/workstation/bin
cp -rf /home/enzo/gap_sdk/tools/ld /home/enzo/gap_sdk/install/workstation
cp -rf /home/enzo/gap_sdk/utils/rules /home/enzo/gap_sdk/install/workstation
cd gap8 && git submodule update --init rtos/pulp/pulp-os
make -C /home/enzo/gap_sdk/rtos/pmsis/api -f tools/export.mk build
make[2]: Entering directory '/home/enzo/gap_sdk/rtos/pmsis/api'
make[2]: Nothing to be done for 'build'.
make[2]: Leaving directory '/home/enzo/gap_sdk/rtos/pmsis/api'
make -C /home/enzo/gap_sdk/gap8/rtos/pulp build
make[2]: Entering directory '/home/enzo/gap_sdk/gap8/rtos/pulp'
make -C /home/enzo/gap_sdk/utils/gap_configs all_scons INSTALL_DIR=/home/enzo/gap_sdk/install/workstation TARGET_INSTALL_DIR=/home/enzo/gap_sdk/install/GAP8_V3
make[3]: Entering directory '/home/enzo/gap_sdk/utils/gap_configs'
#### Building in /home/enzo/gap_sdk/build
#### Release type is Debug
#### Installing to /home/enzo/gap_sdk/install/workstation
scons -Q
make[3]: scons: Command not found
Makefile:51: recipe for target 'all_scons' failed
make[3]: *** [all_scons] Error 127
make[3]: Leaving directory '/home/enzo/gap_sdk/utils/gap_configs'
Makefile:4: recipe for target 'pulp_configs' failed
make[2]: *** [pulp_configs] Error 2
make[2]: Leaving directory '/home/enzo/gap_sdk/gap8/rtos/pulp'
gap8/gap8.mk:106: recipe for target 'pulp-os' failed
make[1]: *** [pulp-os] Error 2
make[1]: Leaving directory '/home/enzo/gap_sdk'
Makefile:42: recipe for target 'all' failed
make: *** [all] Error 2

sousoux, 2023-06-04:
Do a make clean before and then a make all rather than minimal. The script was not run.

sousoux, 2023-06-04:
Also add the output of `which python` from your command line

enzoniko, 2023-06-04:
The python version was confusing, it was using 2.7 in the scripts, and 3.8 in the command line.
I decided to try one last time on a Ubuntu 20.04 Focal Fossa, this time I verified that python was 3.8 and did all the steps again.
It worked, at least I think it did. Basically the `make minimal` worked and it took a while. Then I had to register for autotiler, and it finished without errors. 
Tomorrow I will try to connect the board and test some examples. Now it should work only by pluggin the board in my computer, correct?
I have other questions:
To run ML algorithms made with Python/TensorFlow, what are the requirements? Do I need to use the `make all` command? or the `make minimal` suffices?  And what is the general procedure to run those algorithms?
Thanks in advance for all the help!

sousoux, 2023-06-05:
Please read the documentation carefully and study the examples.

closed 2023-06-05 by sousoux