GreenWaves-Technologies/gap_sdk #395: Problem with generating the model in coco object detection notebook with nntools

gwgknudayanga, 2023-08-03:
hi,
I have executed the the 'execute_on_target' in the following way. ( i had to remove 'write_out_to_file' and 'cmake' as it always tells those two are not options for execute_on_target function. And i set the source to gapuino_v3 since there was no any gap9_v2.cfg file with the latest master branch for gap_sdk).

res = G.execute_on_target(
pmsis_os='freertos',
platform="gvsoc",
directory="test_run",
input_tensors=int_execution[0],
output_tensors=0,
source='gapuino_v3',
#write_out_to_file=True,
at_log=True,
dont_run=False,
settings={
'l1_size': 128000,
'l2_size': 1300000,
'tensor_directory': './tensors'
},
#cmake=True,
at_loglevel=1,
print_output=True
)
for l in res.at_log[-29:-1]:
print(l)

After executing that i am getting following errors.

home/udayanga/gap_sdk/tools/autotiler_v3/CNN_Libraries_NE16/CNN_BasicKernels_NE16.c:765: undefined reference to hal_cl_eu_event_mask_wait_and_clear' [/home/udayanga/coco_object_local_project/test_run/BUILD/GAP8_V3/GCC_RISCV_FREERTOS//home/udayanga/gap_sdk/tools/autotiler_v3/CNN_Libraries_NE16/CNN_BasicKernels_NE16.o](https://file+.vscode-resource.vscode-cdn.net/home/udayanga/coco_object_local_project/test_run/BUILD/GAP8_V3/GCC_RISCV_FREERTOS//home/udayanga/gap_sdk/tools/autotiler_v3/CNN_Libraries_NE16/CNN_BasicKernels_NE16.o):/home/udayanga/gap_sdk/tools/autotiler_v3/CNN_Libraries_NE16/CNN_BasicKernels_NE16.c:765: more **undefined references to hal_cl_eu_event_mask_wait_and_clear' follow**

collect2: error: ld returned 1 exit status

Expect your help. Thank you.

gwgknudayanga, 2023-08-06:
This worked after setting proper l1 and l2 sizes for gap8. so closing

closed 2023-08-06 by gwgknudayanga

Brittanyzhou, 2024-01-08:
I tried in GAP8 but occurs camera problems and I can't fix it. I want to know how you handle this problem?

gwgknudayanga, 2024-09-15:
@Brittanyzhou, for GAP8, i commented out use_ne16 and hwc = True lines (seems they are only for GAP9.
And then i used the following settings so that l1 l2 memories match for gap8. It needs to reserve l2 memory as low as possible for autotiler because then we can have more memory to load the models,

res = G.execute_on_target(
pmsis_os='freertos',
platform="gvsoc",
directory="test_run",
input_tensors=int_execution[0],
output_tensors=0,
source='gapuino_v3',
#write_out_to_file=True,
at_log=True,
dont_run=False,
settings={
'l1_size': 64000,
'l2_size': 320000,
'tensor_directory': './tensors'
},
#cmake=True,
at_loglevel=1,
print_output=True
)

Also i replaced the 126 line of ~/gap_sdk/tools/nntool/nntool/execution/kernels/quant/ssd_postprocess.py as follows to visulaize the bboxes properly in gap_sdk for gap8.
temp = np.array([bbox['bbox'][0][0],bbox['bbox'][1][0],bbox['bbox'][2][0],bbox['bbox'][3][0]],dtype=np.int32)
 out_boxes[out_idx] = temp

Hope this will help.

Thanks and Rgds,
Udayanga