GreenWaves-Technologies/gap_sdk #398: how to measure the power consumption  and latency in gvsoc simulation

gwgknudayanga, 2023-08-09:
Dear experts,

How to measure the power consumption in running my application for object detection in gvsoc platform?
Thank you.

Thanks and Rgds,
Udayanga

Yaooooo, 2023-08-09:
Hi, 

You cannot really "measure" the power from GVSOC, but you can do an estimation based on the cycles number you measured. 
Because GAP is a DVFS system and you can change the voltage & freq according to the perf you need.
And different voltage & freq correspond to different power consumption. 

Then if you want to have precise power consumption number, the only way is to measure from the chip. 

Regards,
Yao

gwgknudayanga, 2023-08-09:
Hi,
Thanks for your reply. Is there any reasonable value for 'per cycle power'? Because when i want to compare the power values with another (such as jetson ) then i need to get some reasonable estimation for the power of GAP 8. 

Rgds,
Niranjan

Yaooooo, 2023-08-09:
Hi,

No, because as I said, it's a DVFS system, same cycles in different frequency has different power consumption. 
It can be scaled from few hundreds of uW to around 100mW.

Reasonable estimation with GVSOC is: 

1. Understand what you want to run, and how many cycles it takes. For example, if it's a NN. then how many cycles an inference will take.
2. According to your performance requirements, estimate the minimum frequency needed. For example, how many frames per secound.
3. Then according to the frequency, we can give you an rough estimation power consumption.

gwgknudayanga, 2023-08-10:
Hi,

I was running object detection of coco test data'(300x300x3 resolution images) using ssd_mobilenetv2_quant.lite model. For a single inference, it reports 679459826 cycles. Also in running ,it is set the frequencies as '250000000 Hz, CL Frequency = 175000000 Hz, PERIIPH Frequency = 250000000 Hz'. To do a single inference for a single image it takes about 4s (it is too large though).
So could you please explain as an example how the power is estimated in this case? I mean the available APIs from your side and how they are going to be applied for the mentioned inference statistics? 
Thank you.

Rgds,
Niranjan