GreenWaves-Technologies/gap_sdk #400: Graph constructor error

Decol-24, 2023-09-18:
Dear experts,

I am using the Gap8_V3 and the latest version of the SDK.
I followed GPAflow, defined and trained the MobileNetV2 model with pytorch, converted it to onnx, and generated the project via the NNTool Python API.

But when I compile the project by `make clean all run platform=gvsoc PMSIS_OS=pulpos`, I get the error:
```
Graph constructor exited with error: 3
(check the generated file mob_t2Kernels.c to see which memory has failed to be allocated)
make: *** [/home/ihpc/Code/gap/gap_sdk/utils/rules/pulp_rules.mk:258: run] Error 250
```
First I suspected that L2 memory is not enough, so all channels were reduced to 1/4 (inputs are 64*64), but this didn't work.

Then I found out that the problem is in this line of the constructor function:
`mob_t2_L2_Memory_Dyn = (AT_L2_POINTER) AT_L2_ALLOC(0, 262160);`
mob_t2_L2_Memory_Dyn equals 0 and returns an error.
When I removed a few layers from the model the problem was solved, but that's not what I wanted.

I used `AT_L2_ALLOC(0, 1);` placed at the beginning of the constructor to check out the memory allocation.
For projects that compile successfully, the return address is 469858104, while for projects with problems, the value is 469939864.
I think some of the L2 memory is allocated by mistake before constructing the model, but I don't know how to fix it.

In addition, I ran the project on NNTool Python and got the following error:
```
make: *** [/home/ihpc/Code/gap/gap_sdk/utils/rules/pulp_rules.mk:249: image] Error 2
make: *** Waiting for unfinished jobs....
make: Leaving directory '/home/ihpc/Code/0822_gap6/build1'
make: Entering directory '/home/ihpc/Code/0822_gap6/build1'

[/sys/board/chip/soc/efuse/warning] Unable to load stimuli file: efuse_preload.data: No such file or directory
Magick: abort due to signal 6 (SIGABRT) "Abort"...
make: *** [/home/ihpc/Code/gap/gap_sdk/utils/rules/pulp_rules.mk:258: run] Aborted (core dumped)
```
I don't know if that has any relation to the question.

Expect your help. Thank you.

sousoux, 2023-09-18:
Your suspicion is correct. The memory allocated before is for your program and what is left is not enough for the model. Try constraining the autotiler more on L2 memory. Look at the nntool documentation to see how to do this.

closed 2023-09-18 by sousoux

gwgknudayanga, 2024-09-14:
@sousoux how did you fix this issue? i am getting the same error? Could you please tell how do you constrain the tiler on L2 or any other workaround you made to resolve this?
Thank you.

Bst Rgds,
Udayanga