GreenWaves-Technologies/gap_sdk #404: NNTool Execution: ModuleNotFoundError: No module named 'stats'

boomer319, 2023-11-29:
The [stats](https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/tools/nntool/nntool/stats) package is not being found. It seems to be an easy fix: Pointing towards the package. What can I do?

    Traceback (most recent call last):
      File "/home/user/repo/gap_sdk/tools/nntool/scripts/nntool", line 107, in <module>
        main()
      File "/home/user/repo/gap_sdk/tools/nntool/scripts/nntool", line 89, in main
        mod.generate_code(args)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/interpreter/generator.py", line 72, in generate_code
        nntool_shell.load_state_file(args.graph_file)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/interpreter/nntool_shell_base.py", line 375, in load_state_file
        history = json.load(fp, cls=JsonSerializableStateDecoder)
      File "/usr/lib/python3.10/json/__init__.py", line 293, in load
        return loads(fp.read(),
      File "/usr/lib/python3.10/json/__init__.py", line 359, in loads
        return cls(**kw).decode(s)
      File "/usr/lib/python3.10/json/decoder.py", line 337, in decode
        obj, end = self.raw_decode(s, idx=_w(s, 0).end())
      File "/usr/lib/python3.10/json/decoder.py", line 353, in raw_decode
        obj, end = self.scan_once(s, idx)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/utils/json_serializable.py", line 140, in object_hook
        return JsonSerializable.from_dict(obj)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/utils/json_serializable.py", line 52, in from_dict
        json_module = importlib.import_module(val['__module_name'])
      File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
        return _bootstrap._gcd_import(name[level:], package, level)
      File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
      File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
      File "<frozen importlib._bootstrap>", line 992, in _find_and_load_unlocked
      File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
      File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
      File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
      File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
    ModuleNotFoundError: No module named 'stats'

boomer319, 2023-11-29:
I did this:
`export PYTHONPATH=$PYTHONPATH:~/repo/gap_sdk/tools/nntool/nntool`

It found stats but raised errors:

    Traceback (most recent call last):
      File "/home/user/repo/gap_sdk/tools/nntool/scripts/nntool", line 107, in <module>
        main()
      File "/home/user/repo/gap_sdk/tools/nntool/scripts/nntool", line 89, in main
        mod.generate_code(args)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/interpreter/generator.py", line 72, in generate_code
        nntool_shell.load_state_file(args.graph_file)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/interpreter/nntool_shell_base.py", line 383, in load_state_file
        self.replay_history()
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/interpreter/nntool_shell_base.py", line 365, in replay_history
        if self.onecmd_plus_hooks(command, add_to_history=False):
      File "/home/aurelio/environments/GAP/lib/python3.10/site-packages/cmd2/cmd2.py", line 1695, in onecmd_plus_hooks
        self.pexcept(ex)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/interpreter/nntool_shell_base.py", line 357, in pexcept
        raise msg
      File "/home/aurelio/environments/GAP/lib/python3.10/site-packages/cmd2/cmd2.py", line 1661, in onecmd_plus_hooks
        stop = self.onecmd(statement, add_to_history=add_to_history)
      File "/home/aurelio/environments/GAP/lib/python3.10/site-packages/cmd2/cmd2.py", line 2081, in onecmd
        stop = func(statement)
      File "/home/aurelio/environments/GAP/lib/python3.10/site-packages/cmd2/decorators.py", line 223, in cmd_wrapper
        return func(cmd2_app, args)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/interpreter/commands/fquant.py", line 113, in do_fquant
        quantizer.quantize()
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/quantizer/new_quantizer.py", line 975, in quantize
        self.bfs_pass()
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/quantizer/new_quantizer.py", line 651, in bfs_pass
        self._bfs_pass(self._graph, node, self._qset, visited)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/quantizer/new_quantizer.py", line 587, in _bfs_pass
        self._bfs_pass(cur_G, edge.to_node, qset,
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/quantizer/new_quantizer.py", line 553, in _bfs_pass
        qrec = self._resolve_qrec(
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/quantizer/new_quantizer.py", line 613, in _resolve_qrec
        qrec = self._call_quantizer(cur_G,
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/quantizer/new_quantizer.py", line 448, in _call_quantizer
        return handler.quantize(node, in_qs, stats, **kwargs)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/unified_quantization_handler.py", line 123, in quantize
        return cls._quantize(params, in_qs, stats, **kwargs)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/multiplicative/quantizers/filter_mult.py", line 300, in _quantize
        return cls._quantize_sw(params, in_qs, stats, np.int8, **kwargs)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/multiplicative/quantizers/filter_mult.py", line 225, in _quantize_sw
        min_val, max_val = stats.get_range_out(0, bits=out_bits)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/range_provider.py", line 50, in get_range_out
        return self._provider.get_range(self._nid, 'out', idx, bits=bits, dtype=dtype,
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/range_provider.py", line 195, in get_range
        return self.get_range_out(nid, idx, bits=bits, dtype=dtype)
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/range_provider.py", line 170, in get_range_out
        stat = self.get_stat(nid)['range_out'][idx]
      File "/home/user/repo/gap_sdk/tools/nntool/nntool/quantization/range_provider.py", line 135, in get_stat
        raise RangesMissingError(f'no ranges found for {nid}')
    nntool.quantization.range_provider.RangesMissingError: no ranges found for DEPTHWISE_CONV_2D_0_128

sousoux, 2023-11-30:
Did you source the sdk?