GreenWaves-Technologies/gap_sdk #408: Cannot connect to the server while make Autotiler

ChenyuZhaoTHU, 2024-01-19:
I am doing make Autotiler command, but after I enter the url, it can not keep going. How can I fix it?

##########################
Do you agree with the license (y/n) ? 

Do you agree with the license (y/n) ? 

Do you agree with the license (y/n) ? y
--2024-01-19 09:46:56--  https://greenwaves-technologies.com/autotiler/libtile.4.2.0.a
Resolving greenwaves-technologies.com (greenwaves-technologies.com)... 35.197.241.59
Connecting to greenwaves-technologies.com (greenwaves-technologies.com)|35.197.241.59|:443... connected.
HTTP request sent, awaiting response... 307 Temporary Redirect
Location: https://gwt-website-files.s3.eu-central-1.amazonaws.com/libtile.4.2.0.a [following]
--2024-01-19 09:46:57--  https://gwt-website-files.s3.eu-central-1.amazonaws.com/libtile.4.2.0.a
Resolving gwt-website-files.s3.eu-central-1.amazonaws.com (gwt-website-files.s3.eu-central-1.amazonaws.com)... 157.240.7.8, 2a03:2880:f12d:83:face:b00c:0:25de
Connecting to gwt-website-files.s3.eu-central-1.amazonaws.com (gwt-website-files.s3.eu-central-1.amazonaws.com)|157.240.7.8|:443... connected.
ERROR: no certificate subject alternative name matches
        requested host name 'gwt-website-files.s3.eu-central-1.amazonaws.com'.
To connect to gwt-website-files.s3.eu-central-1.amazonaws.com insecurely, use `--no-check-certificate'.
make[2]: *** [Makefile:23: Autotiler/LibTile.a] Error 1
make[2]: Leaving directory '/gap_sdk/tools/autotiler_v3'
make[1]: *** [gap8/gap8.mk:258: autotiler] Error 2
make[1]: Leaving directory '/gap_sdk'
make: *** [Makefile:42: all] Error 2

ChenyuZhaoTHU, 2024-01-19:
I found I can download the file by click the link https://greenwaves-technologies.com/autotiler/libtile.4.2.0.a. Can I add this file manually?

gwtsivasiva, 2024-01-19:
Can you provide your email address?

Yao Zhang
Greenwaves Technologies

ChenyuZhaoTHU, 2024-01-19:
Hi Yao,Here is my email ***@***.*** you!

Yaooooo, 2024-01-19:
Seems the email is hided by github, can you send it to the contact via our website?

ChenyuZhaoTHU, 2024-01-19:
OK. But I don’t understand which website you refers to.

Yaooooo, 2024-01-19:
We don't have other websites, but only greenwaves-technologies.com

-- 

*Yao ZHANG*Global Customer Success Director
APAC Business Development Director
Tel: 06.82.78.41.71
Mail: ***@***.***
Skype: zhyao930
Wechat: 453115693
-------------------------------------------------------------------------------------
[image: https://greenwaves-technologies.com/]
<https://greenwaves-technologies.com/>

alvinxds, 2024-02-28:
hello , Ialso have a similar issue.Do you agree with the license (y/n) ? 

Do you agree with the license (y/n) ? y
--2024-02-28 17:52:04--  https://greenwaves-technologies.com/autotiler/libtile.4.2.0.a
Resolving greenwaves-technologies.com (greenwaves-technologies.com)... 35.197.241.59
Connecting to greenwaves-technologies.com (greenwaves-technologies.com)|35.197.241.59|:443... connected.
HTTP request sent, awaiting response... 307 Temporary Redirect
Location: https://gwt-website-files.s3.eu-central-1.amazonaws.com/libtile.4.2.0.a [following]
--2024-02-28 17:52:04--  https://gwt-website-files.s3.eu-central-1.amazonaws.com/libtile.4.2.0.a
Resolving gwt-website-files.s3.eu-central-1.amazonaws.com (gwt-website-files.s3.eu-central-1.amazonaws.com)... 31.13.69.245, 2a03:2880:f12a:83:face:b00c:0:25de
Connecting to gwt-website-files.s3.eu-central-1.amazonaws.com (gwt-website-files.s3.eu-central-1.amazonaws.com)|31.13.69.245|:443... failed: Connection refused.
Connecting to gwt-website-files.s3.eu-central-1.amazonaws.com (gwt-website-files.s3.eu-central-1.amazonaws.com)|2a03:2880:f12a:83:face:b00c:0:25de|:443... failed: Cannot assign requested address.
Retrying.

--2024-02-28 17:53:12--  (try: 2)  https://gwt-website-files.s3.eu-central-1.amazonaws.com/libtile.4.2.0.a
Connecting to gwt-website-files.s3.eu-central-1.amazonaws.com (gwt-website-files.s3.eu-central-1.amazonaws.com)|31.13.69.245|:443... failed: Connection refused.
Connecting to gwt-website-files.s3.eu-central-1.amazonaws.com (gwt-website-files.s3.eu-central-1.amazonaws.com)|2a03:2880:f12a:83:face:b00c:0:25de|:443... failed: Cannot assign requested address.
Retrying.

--2024-02-28 17:53:16--  (try: 3)  https://gwt-website-files.s3.eu-central-1.amazonaws.com/libtile.4.2.0.a
Connecting to gwt-website-files.s3.eu-central-1.amazonaws.com (gwt-website-files.s3.eu-central-1.amazonaws.com)|31.13.69.245|:443... failed: Connection timed out.
Connecting to gwt-website-files.s3.eu-central-1.amazonaws.com (gwt-website-files.s3.eu-central-1.amazonaws.com)|2a03:2880:f12a:83:face:b00c:0:25de|:443... failed: Cannot assign requested address.

It keeps retrying. I also got the URL and pasted it in the appropriate place. Please help me out, 
Thanks!

gwtsivasiva, 2024-02-28:
Hello,

Since the file is hosted on Amazon, it may not be accessible for some
regions.
If it's the case, please send us your request by filling the "contact" on
our website.

Best regards.

-- 

*Yao ZHANG*Global Customer Success Director
APAC Business Development Director
Tel: 06.82.78.41.71
Mail: ***@***.***
Skype: zhyao930
Wechat: 453115693
-------------------------------------------------------------------------------------
[image: https://greenwaves-technologies.com/]
<https://greenwaves-technologies.com/>