GreenWaves-Technologies/gap_sdk #412: ValueError: Cannot generate strided slice layer _model_29_Slice_3 with a copy if start != 0

Jool99, 2024-03-15:
Getting the error as in the title.

Below is the node in Netron.

All help is appreciated. Thanks in advance!

[image]

[image]

Jool99, 2024-03-15:
I solved it by replacing the two onnx slices with one split.

sousoux, 2024-03-15:
If you have a graph that you can share that reproduces this we would like to look at it. It should be automatically replaced with a split by nntool.

Jool99, 2024-03-15:
Here they are:

[slices_vs_split.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/14618126/slices_vs_split.zip)

You can find the difference by searching for the only two slices in with_slices.onnx.