GreenWaves-Technologies/gap_sdk #413: error: storage size of 'default_flash_conf' isn't known

Jool99, 2024-03-18:
When trying to do "make clean model build image" in the bitcraze ai deck [classification example](https://github.com/bitcraze/aideck-gap8-examples/tree/master/examples/ai/classification), I get the following:

```
gap_sdk/tools/autotiler_v3/Emulation/at_api_pmsis.h: In function '__at_default_flash_fs_open':
gap_sdk/tools/autotiler_v3/Emulation/at_api_pmsis.h:206:32: error: storage size of 'default_flash_conf' isn't known
   struct pi_default_flash_conf default_flash_conf;
                                ^~~~~~~~~~~~~~~~~~
```

I am using the latest v4.22.0 gap_sdk on Ubuntu 22.04.4.

Jool99, 2024-03-18:
Edit: Initially wrote about another example but it is easier to reproduce with this one.

Yaooooo, 2024-03-25:
Hi

This is only related to how the BSP is configured. 
This type has been only defined in these 3 BSP: 
yao@yao ~/projects/sdk/gap8_sdk/rtos/pmsis/bsp (master)$grep -rnIw "pi_default_flash_conf"

include/bsp/gapoc_b.h:82:#define pi_default_flash_conf pi_hyperflash_conf
include/bsp/gapoc_b_v2.h:88:#define pi_default_flash_conf pi_hyperflash_conf
include/bsp/gapuino.h:99:#define pi_default_flash_conf pi_hyperflash_conf

You can add it in ai_deck if you need.

closed 2024-04-08 by Jool99