GreenWaves-Technologies/gap_sdk #416: Unable to run helloworld on GAP9_EVK 3.1

Alejandroperezm, 2024-05-27:
Good morning,

 I Have succesfully installed the SDK, however whenever I tried to run the helloworld I get this error that im unabled to solve. It can´t find GAP9 module.

`[100%] Linking C executable helloworld
[100%] Built target helloworld
Traceback (most recent call last):
  File "/home/osboxes/gap9_sdk_public/utils/gapy_v2/bin/gapylib/target.py", line 53, in get_target
    module = importlib.import_module(target)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 992, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'gap9'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/osboxes/gap9_sdk_public/utils/gapy_v2/bin/gapy", line 92, in <module>
    target = gapylib.target.get_target(target_name)(
  File "/home/osboxes/gap9_sdk_public/utils/gapy_v2/bin/gapylib/target.py", line 55, in get_target
    raise RuntimeError(f'Invalid target specified: {target}') from exc
RuntimeError: Invalid target specified: gap9.evk
gmake[3]: *** [CMakeFiles/image.dir/build.make:70: CMakeFiles/image] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:655: CMakeFiles/image.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:714: CMakeFiles/run.dir/rule] Error 2
gmake: *** [Makefile:280: run] Error 2`

I have the GAP9 EVK v3.1 board connected by USB to PC, the FTDI controller RS232SH is recogniced by ubuntu (4 new USBs appear in dev/tty*). Boot pins from J10 and J9 are left open.

Moreover, in the SDK configuration Im not able to select GAPMOD 2.0 or GAP9 EVKv3.1, which im currently using, might this have something to do with it?

Thank you.

Yaooooo, 2024-05-27:
You are using the wrong SDK, this is the SDK for gap8
For GAP9, I have opened you the access for : 
https://github.com/GreenWaves-Technologies/gap9_sdk_public

You should not mix them.

Alejandroperezm, 2024-05-27:
Im actually using GAP9_SDK_PUBLIC, but when selecting target with source sourceme.sh the only target on display is gap9_evk, GAP9_v2. When going thought the steps to build hellowork, in menuconfig, in board selection there are only GAP9mod v1.0b Evaluation kit v1.3 and v2.0 while im using the EVK v3.1 with the GAP9mod v2.0. The project builds but is not able to run due  I to the error showcased previously.

 The board is connected to PC using a  USB 3.0 cable, Ubuntu 22.04 is running on Oracle Virtual Box and does recognize the FTDI Quad RS232 HS. 

Excuse my ignorance, Im not sure if the problem im having with running the helloworld code is due to the SDK not having the correct target gap9 board or something else such as USB port configuration in Linux.