GreenWaves-Technologies/gap_sdk #427: I encountered the following problem when trying to Build and run on GVSoC after Use DORY to generate the C code.

zyxzxy, 2024-10-28:
Hi，
I encountered the following problem when trying to Build and run on GVSoC after Use DORY to generate the C code.

DORY_network/src/main.c:42:10: fatal error: tools/frame_streamer.h: No such file or directory
#include "tools/frame_streamer.h"
^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [/home/zyx/gap_gnu_toolchain/gap_gnu_toolchain/gap_gnu_toolchain/gap8_openocd/gap_sdk/tools/rules/pulp_rules.mk:205: /home/zyx/下载/pulp-dronet-master/tiny-pulp-dronet-v3/drone-applications/gap8-dronet-app/pulp-dronet-v3/BUILD/GAP8_V2/GCC_RISCV/DORY_network/src/main.o] Error 1

My gap sdk version is GAP SDK 3.9.1.After I found the frame_streamer.h file and imported it, I encountered a new problem.

/usr/lib/gap_riscv_toolchain/bin/../lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: cannot find -lgaptools
collect2: error: ld returned 1 exit status
make: *** [/home/zyx/gap_gnu_toolchain/gap_gnu_toolchain/gap_gnu_toolchain/gap8_openocd/gap_sdk/tools/rules/pulp_rules.mk:212: /home/zyx/下载/pulp-dronet-master/tiny-pulp-dronet-v3/drone-applications/gap8-dronet-app/pulp-dronet-v3/BUILD/GAP8_V2/GCC_RISCV/main] Error 1
How can I solve these problems?
Best wishes！
ZHANG

Yaooooo, 2024-10-28:
I believe you mean Dory from PULP, right? 
We don't support Dory in our SDK, please check with PULP.

zyxzxy, 2024-10-28:
感谢您的回复，我几天前也询问了pulp的开发人员，他们表示曾经也在这里问过这个问题，好像也是一年前您回复的。在他们的论文里是使用的dory并部署在AI-deck上了，不知道他们是怎么解决的。。

gwtsivasiva, 2024-10-28:
这个真的帮不上忙了，这边没人玩过Dory。我记得这个项目他们好像用Dory跑的：
https://github.com/pulp-platform/pulp-dronet

yandeng0718, 2025-03-12:
想问下你部署成功了吗，我最近也在跑pulp-dronet-v3的代码

zyx77777, 2025-03-12:
> 想问下你部署成功了吗，我最近也在跑pulp-dronet-v3的代码

并没有成功，而且我现在自己改了个分类网络也遇到了问题