GreenWaves-Technologies/gap_sdk #433: Site Down and Missing File for Autotiler Setup

SimoMila91, 2025-03-14:
Hello, 

I have encountered two issues trying to set up the GAP8 SDK using the Docker container provided (**bitcraze/aideck**) to work with the AI Deck on my Crazyflie. However, I am encountering a critical issue: the entire GreenWaves Technologies website (https://greenwaves-technologies.com) appears to be down and returns a 404 Not Found error.

This prevents me from downloading the required libtile.4.3.5.a file during the make autotiler step. Here are the details:

Steps to reproduce:

Run the Docker container:
`docker run --rm -it --name myAiDeckContainer bitcraze/aideck`

Navigate to the GAP8 SDK directory and source the configuration:
```
cd /gap_sdk
source configs/ai_deck.sh
```

Run make autotiler:
`make autotiler`
Accept the licence terms when prompted.

The script attempts to download libtile.4.3.5.a from:
`https://greenwaves-technologies.com/autotiler/libtile.4.3.5.a
`
However, the entire domain is returning a 404 error. 

Are there alternative sources for the the libtile.4.3.5.a file? 
Could you let us know when the website might be back online? 

Thank you for ur support.

Valeredeterre, 2025-03-18:
It appears that they are going bankrupt.

https://annuaire-entreprises.data.gouv.fr/annonces/808076582

paulxiongfree, 2025-03-27:
> It appears that they are going bankrupt.
> 
> https://annuaire-entreprises.data.gouv.fr/annonces/808076582

This is really sad news.

ignatious-p, 2025-04-03:
If anyone could share LibTile.a online that would help out a lot.